riscvarchive/riscv-binutils-gdb #2: Build error:

cliffordwolf, 2015-11-04:
Trying to build this package I get the following error:

    gcc -DHAVE_CONFIG_H   -DWITH_DEFAULT_MODEL='"RV64G"'  -DPROFILE=1 -DWITH_PROFILE=-1   -DWITH_ALIGNMENT=NONSTRICT_ALIGNMENT -DWITH_TARGET_WORD_BITSIZE=64 -DWITH_TARGET_BYTE_ORDER=LITTLE_ENDIAN -DWITH_ENVIRONMENT=ALL_ENVIRONMENT   -DWITH_HOST_BYTE_ORDER=LITTLE_ENDIAN        -Wall -Wdeclaration-after-statement -Wpointer-arith -Wpointer-sign -Wno-unused -Wunused-value -Wunused-function -Wno-switch -Wno-char-subscripts -Wmissing-prototypes -Wdeclaration-after-statement -Wempty-body -Wmissing-parameter-type -Wold-style-declaration -Wold-style-definition -Wformat-nonliteral       -I. -I../../../sim/riscv -I../common -I../../../sim/riscv/../common -I../../include -I../../../sim/riscv/../../include -I../../bfd -I../../../sim/riscv/../../bfd -I../../opcodes -I../../../sim/riscv/../../opcodes  -g -O2 -c -o sim-model.o -MT sim-model.o -MMD -MP -MF .deps/sim-model.Tpo ../../../sim/riscv/../common/sim-model.c
    ../../../sim/riscv/../common/sim-model.c: In function ‘model_set’:
    ../../../sim/riscv/../common/sim-model.c:108:3: warning: implicit declaration of function ‘CPU_MACH’ [-Wimplicit-function-declaration]
       CPU_MACH (cpu) = MODEL_MACH (model);
       ^
    ../../../sim/riscv/../common/sim-model.c:108:18: error: lvalue required as left operand of assignment
       CPU_MACH (cpu) = MODEL_MACH (model);
                      ^
    ../../../sim/riscv/../common/sim-model.c:109:3: warning: implicit declaration of function ‘CPU_MODEL’ [-Wimplicit-function-declaration]
       CPU_MODEL (cpu) = model;
       ^
    ../../../sim/riscv/../common/sim-model.c:109:19: error: lvalue required as left operand of assignment
       CPU_MODEL (cpu) = model;
                       ^
    [....]

I'm trying to build using the following sequence of commands (same with  `--target=riscv32`):

    mkdir build
    cd build/
    ../configure --program-prefix=riscv- --target=riscv
    make

Also: How does this package relate to https://github.com/mythdraenor/riscv-gdb? The webpage (http://riscv.org/download.html#tab_gdb) points to that repo, but it has not seen an update in over a year..

palmer-dabbelt, 2015-11-09:
I just built the latest master on Ubuntu 14.04 and I don't get any errors.  sim/common/cgen-types.h makes it look like SIM_HAVE_MODEL is always defined, which sim/common/sim-cpu.h uses to define CPU_MACH and CPU_MODEL.

Can you try some standard stuff like doing a clean clone and build?  If that doesn't work, can you give me some details of your setup so I can try and actually fix this?

I believe this is the correct repository to point to, not riscv-gdb.

palmer-dabbelt, 2015-11-09:
I'm trying to get a version of our stuff rebased on upstream's HEAD, and I'm seeing these errors.  I'm not sure why you're getting them and I'm not, but with any luck I'll end up fixing this.

palmer-dabbelt, 2015-11-09:
I don't know what github is doing with the issues, but how does this look

````
commit 3d29d2a30eb2efd30f8d01a0f4811a3438ebffee
Author: Palmer Dabbelt <palmer@dabbelt.com>
Date:   Sun Nov 8 23:49:42 2015 -0800

    Fix RISC-V sim for new binutils/gdb versions

    I'm not sure when or why this broke, or why adding tconfig.h fixes it, but it
    appears to.  Note that if I add tconfig.h and don't do a clean build then I get
    some linking errors.

    Pretty much all of this came from the bfin sim port, as that looks most similar
    to ours.

diff --git a/sim/riscv/machs.c b/sim/riscv/machs.c
index ef08319..c968109 100644
--- a/sim/riscv/machs.c
+++ b/sim/riscv/machs.c
@@ -21,6 +21,10 @@
 #include "config.h"

 #include "sim-main.h"
+#include "gdb/sim-riscv.h"
+#include "bfd.h"
+
+#include "sim-hw.h"

 static void
 riscv_model_init (SIM_CPU *cpu)
diff --git a/sim/riscv/tconfig.h b/sim/riscv/tconfig.h
new file mode 100644
index 0000000..b109533
--- /dev/null
+++ b/sim/riscv/tconfig.h
@@ -0,0 +1,2 @@
+/* This was copied from bfin, and is listed as a hack there.  */
+#define SIM_HAVE_MODEL
````

cliffordwolf, 2015-11-09:
I've checked out `riscv-rebase-2015.10.08` (3d29d2a) from your git repo and I can successfully build gdb from it using the following commands:

    mkdir build && cd build
    ../configure --target=riscv32-unknown-elf --prefix=/opt/riscv32i
    make -j4 && make install

Thanks!

Maybe you can also point me in the right direction or refer me to the right person to ask regarding the following issue:

I want to write a gdbserver for PicoRV32 (going to implement a hw debug interface for it). According to [the gdb docs](https://sourceware.org/gdb/onlinedocs/gdb/Overview.html#Overview) only the following packets need to be implemented for a minimal gdb server: `g, G, m, M, c, s`. However, when the gdb I just built connects to my dummy gdb server I just wrote I see the following requests sent to by gdb:

    $qSupported:multiprocess+;swbreak+;hwbreak+;qRelocInsn+;vContSupported+#ff
    $Hg0#df
    $qTStatus#49
    $?#3f
    $qfThreadInfo#bb
    $qL1200000000000000000#50
    $Hc-1#09
    $qC#b4
    $qAttached#8f

(Note that none of them is in `g, G, m, M, c, s`.)

I respond to all of them by sending `+`, followed by `$#00`, as this is the respond that I should send for an unsupported command, according to the docs as I understand them. After that sequence of pkts sent to my gdb server, gdb just hangs. This is the output I see:

    (gdb) target remote :1234
    Remote debugging using :1234
    warning: Invalid remote reply: 

Is there an existing simple gdb stub (preferably for RISC-V) that I could use as a reference? Any other useful hints that might help me?

vapier, 2015-11-24:
you probably want to use the sw-dev mailing list for questions like this

closed 2015-11-24 by vapier