riscvarchive/riscv-binutils-gdb #7: riscv-gdb-7.10 : info locals get function variable values by using sp instead of fp (s0)

Dolu1990, 2016-04-19:
Hi,

For example, when GDB want to probe the value of a function variable at (for example) s0-28, it does the access by using sp-28, which return the wrong value.

Dolu1990, 2016-06-05:
nobody tried ?

palmer-dabbelt, 2016-07-12:
Sorry, I didn't look at this because I got too busy.  Can you provide an actual test case.  Ideally I'd like:

* What command you ran
* What output you got
* What output you want

If you have a patch that would be great, otherwise I'll try to look at it in a bit.

Dolu1990, 2016-07-13:
Hi,
I'm currently in holiday away from my PC, but from my remembers : 
- riscv gdb 32 bit
- command -> info locals
- bug -> get function's local variables by using the stack in place of using the frame pointer ( as the binary do )
int foo(int arg){
  int a,b,c;
  a = 2;
  b = 5;
  c = 7;
  return a + b + c + arg; //break here
}

Then if i info locals, a b c are readed at the wrong address by GDB.

palmer-dabbelt, 2017-03-06:
I believe this has been fixed.

closed 2017-03-06 by palmer-dabbelt

Dolu1990, 2017-06-14:
Yes ^^
Now it work fine :D