riscvarchive/riscv-binutils-gdb #8: can't use some gdb command such as set $pc, info registers?

lx324310, 2016-04-26:
Hi:
I have built the RISC-V GDB tool from this git 
here is my steps to build it:
git clone https://github.com/riscv/riscv-binutils-gdb.git
cd riscv-binutils-gdb
mkdir build
cd build
../configure --target=riscv32-unknown-elf --prefix=/opt/riscv32gdb
make
make install
after doing this, I used gdb to connect with advance debug, the result as follows :

(gdb) target remote :1234
Remote debugging using :1234
(gdb) load
Loading section .vectors, size 0x8c lma 0x0
Loading section .text, size 0x220 lma 0x8c
Loading section .text.startup.main, size 0x4c lma 0x2ac
Loading section .text.illegal_insn_handler_c, size 0x2c lma 0x2f8
Loading section .text.qprinti.constprop.2, size 0x3e4 lma 0x324
Loading section .text.printf, size 0x7a0 lma 0x708
Loading section .text.puts, size 0x60 lma 0xea8
Loading section .text.set_pin_function, size 0x48 lma 0xf08
Loading section .text.set_gpio_pin_direction, size 0x68 lma 0xf50
Loading section .text.set_gpio_pin_value, size 0x68 lma 0xfb8
Loading section .text.int_time_cmp, size 0x4 lma 0x1020
Loading section .text.int_main, size 0x58 lma 0x1024
Loading section .text.uart_set_cfg, size 0x50 lma 0x107c
Loading section .text.uart_sendchar, size 0x1c lma 0x10cc
Loading section .text.uart_wait_tx_done, size 0x14 lma 0x10e8
Loading section .text.eoc, size 0x60 lma 0x10fc
Loading section .text.exit, size 0x10 lma 0x115c
Loading section .rodata, size 0x1b8 lma 0x100000
Loading section .eh_frame, size 0x190 lma 0x1001b8
Loading section .data, size 0x10 lma 0x100348
Loading section .sdata.setme, size 0x4 lma 0x100358
Loading section .sdata.setme2, size 0x4 lma 0x10035c
Start address 0x8c, load size 5324
Transfer rate: 1 KB/sec, 242 bytes/write.
(gdb) info registers
**The current thread has terminated**
(gdb)

I can load code to memory,but i can't use other gdb command,such as info,set $pc ....

Dolu1990, 2016-04-29:
Hi, 
which branche are you using ? 
With riscv-gdb-7.10 this is fine for me, but i'm using a openOCD fork, not advance debug.

palmer-dabbelt, 2017-03-06:
I'm closing this as it's inactive and I managed to do a "info registers" today.

closed 2017-03-06 by palmer-dabbelt