riscvarchive/riscv-binutils-gdb #19: gdb fails to disassemble RV64 instructions

sorear, 2016-09-01:
Doing experiments on the version of gdb in the Fedora RISC-V porting project.  It's conceivable these are packaging bugs; let me know if you need any additional information.

Exhibit A, the main function in /bin/ls as disassembled by objdump:

```
bash-4.3# objdump --version                       
GNU objdump (GNU Binutils) 2.26.0.20160516
Copyright (C) 2015 Free Software Foundation, Inc.
This program is free software; you may redistribute it under the terms of
the GNU General Public License version 3 or (at your option) any later version.
This program has absolutely no warranty.
bash-4.3# objdump -d /bin/ls | grep -A80 'main>:'
000000000001b0c0 <main>:
   1b0c0:       cc010113                addi    sp,sp,-832
   1b0c4:       32813823                sd      s0,816(sp)
   1b0c8:       00050413                mv      s0,a0
   1b0cc:       0005b503                ld      a0,0(a1)
   1b0d0:       32113c23                sd      ra,824(sp)
   1b0d4:       32913423                sd      s1,808(sp)
   1b0d8:       33213023                sd      s2,800(sp)
   1b0dc:       31313c23                sd      s3,792(sp)
   1b0e0:       2fa13023                sd      s10,736(sp)
   1b0e4:       00058913                mv      s2,a1
   1b0e8:       31413823                sd      s4,784(sp)
   1b0ec:       31513423                sd      s5,776(sp)
   1b0f0:       31613023                sd      s6,768(sp)
   1b0f4:       2f713c23                sd      s7,760(sp)
   1b0f8:       2f813823                sd      s8,752(sp)
   1b0fc:       2f913423                sd      s9,744(sp)
   1b100:       2db13c23                sd      s11,728(sp)
   1b104:       00017d37                lui     s10,0x17
   1b108:       1710a0ef                jal     25a78 <set_program_name>
   1b10c:       968d0593                addi    a1,s10,-1688 # 16968 <__mon_yday+0x3210>
   1b110:       00600513                li      a0,6
   1b114:       f1dff0ef                jal     1b030 <setlocale@plt>
   1b118:       000154b7                lui     s1,0x15
   1b11c:       000155b7                lui     a1,0x15
   1b120:       6d058593                addi    a1,a1,1744 # 156d0 <__mon_yday+0x1f78>
   1b124:       6e848513                addi    a0,s1,1768 # 156e8 <__mon_yday+0x1f90>
   1b128:       f49ff0ef                jal     1b070 <bindtextdomain@plt>
   1b12c:       6e848513                addi    a0,s1,1768
   1b130:       bc1ff0ef                jal     1acf0 <textdomain@plt>
   1b134:       00023537                lui     a0,0x23
   1b138:       00200993                li      s3,2
   1b13c:       9cc50513                addi    a0,a0,-1588 # 229cc <close_stdout>
   1b140:       8b31a823                sw      s3,-1872(gp) # 16af8 <exit_failure>
   1b144:       79d130ef                jal     2f0e0 <atexit>
   1b148:       89c1a483                lw      s1,-1892(gp) # 16ae4 <ls_mode>
   1b14c:       00018737                lui     a4,0x18
   1b150:       d0070713                addi    a4,a4,-768 # 17d00 <rpl_optarg+0x400>
   1b154:       fff00793                li      a5,-1
   1b158:       00070613                mv      a2,a4
   1b15c:       9201a423                sw      zero,-1752(gp) # 16b70 <exit_status>
   1b160:       00100693                li      a3,1
   1b164:       02e13023                sd      a4,32(sp)
   1b168:       94d18423                sb      a3,-1720(gp) # 16b90 <print_dir_name>
   1b16c:       03f79713                slli    a4,a5,0x3f
   1b170:       98f63023                sd      a5,-1664(a2)
   1b174:       9e01bc23                sd      zero,-1544(gp) # 16c40 <pending_dirs>
   1b178:       96e63c23                sd      a4,-1672(a2)
   1b17c:       96018423                sb      zero,-1688(gp) # 16bb0 <qmark_funny_chars>
   1b180:       77348063                beq     s1,s3,1b8e0 <main+0x820>
   1b184:       00300a13                li      s4,3
   1b188:       75448263                beq     s1,s4,1b8cc <main+0x80c>
   1b18c:       00100793                li      a5,1
   1b190:       00f48463                beq     s1,a5,1b198 <main+0xd8>
   1b194:       e8dff0ef                jal     1b020 <abort@plt>
   1b198:       00048513                mv      a0,s1
   1b19c:       cb5ff0ef                jal     1ae50 <isatty@plt>
   1b1a0:       00050463                beqz    a0,1b1a8 <main+0xe8>
[... 101 lines cut ...]
   0x1b1a0 <main+224>:  beqz    a0,0x1b1a8 <main+232>
   0x1b1a4 <main+228>:  j       0x1c878 <main+6072>
   0x1b1a8 <main+232>:  sw      s1,-1592(gp)
   0x1b1ac <main+236>:  sw      zero,-1596(gp)
   0x1b1b0 <main+240>:  sw      zero,-1600(gp)
   0x1b1b4 <main+244>:  sb      zero,-1601(gp)
   0x1b1b8 <main+248>:  sb      zero,-1603(gp)
   0x1b1bc <main+252>:  sb      zero,-1604(gp)
   0x1b1c0 <main+256>:  sb      zero,-1652(gp)
   0x1b1c4 <main+260>:  li      a5,1
   0x1b1c8 <main+264>:  sw      a5,-1656(gp)
   0x1b1cc <main+268>:  sb      zero,-1658(gp)
   0x1b1d0 <main+272>:  sb      zero,-1659(gp)
   0x1b1d4 <main+276>:  0x9601bc23
   0x1b1d8 <main+280>:  lui     a0,0x15
   0x1b1dc <main+284>:  0x9601b823
   0x1b1e0 <main+288>:  addi    a0,a0,1784
   0x1b1e4 <main+292>:  sw      zero,-1628(gp)
   0x1b1e8 <main+296>:  sw      zero,-1664(gp)
   0x1b1ec <main+300>:  sb      zero,-1547(gp)
   0x1b1f0 <main+304>:  jal     0x1ad70 <getenv@plt>
   0x1b1f4 <main+308>:  mv      s1,a0
   0x1b1f8 <main+312>:  beqz    a0,0x1b238 <main+376>
   0x1b1fc <main+316>:  lui     s3,0x13
```

aswaterman, 2016-11-13:
When you get a chance, can you confirm that 9a8cb24 fixes gdb?

sorear, 2016-11-13:
Thanks!  I'm actually not sure right now how to compile gdb myself — when I get that working I'll get back to you.

sorear, 2016-12-16:
@aswaterman This is half fixed in 1cc4048 (it looks like there was a big rebase earlier and I can't easily tell what has been changed since that commit).  Specifically, `set arch riscv:rv64` now works, but gdb still isn't automatically picking up the bitness from the ELF file.  (I ran gdb against a ld.so file, then did `disas _start`)

aswaterman, 2016-12-16:
Thanks for the update. I don't have the time to look into the ELF class detection issue; ideally one of the people working on GDB could do so.

jim-wilson, 2017-11-30:
This has had no comment for almost a year.  I am unable to reproduce the gdb problem using a toolchain built today.

Bisecting on the riscv-gnu-toolchain project narrowed the fix down between these two commits
commit 5dac17852d746f6ea26736a7cdbd073b201ce08f
Author: Palmer Dabbelt <palmer@dabbelt.com>
Date:   Wed May 3 23:56:31 2017 -0700
and
commit 19ddd4d6f42bddaa29a76f926fea9ee5be177766
Merge: ef9bf99 43a55a8
Author: Palmer Dabbelt <palmer@dabbelt.com>
Date:   Wed May 3 22:36:55 2017 -0700

Then looking at riscv-binutils-gdb git log the fix was obvious.
commit d6f218e1bbfea00a1481dad56b702a909474c62f
Author: Kuan-Lin Chen <rufus@andestech.com>
Date:   Tue Jan 17 09:46:28 2017 +0800

    RISC-V/bfd: Hook elf_backend_object_p to set the mach type.

I did builds before and after to verify that this commit fixes it.  So closing.

closed 2017-11-30 by jim-wilson