riscvarchive/riscv-binutils-gdb #22: break dependency on Linux

sbeamer, 2016-09-13:
The riscv simulator depends on Linux's clock_gettime, which prevents this from being compiled on any other platform such as OS X, even if gcc is used. Since this repo is a dependency of riscv-gnu-toolchain, this restriction makes any other OS incapable of using riscv-gcc.

clock_gettime is used to implement CSR_TIME (in sim/riscv/sim-main.c), but it seems any timing mechanism could also be used. Some possibilities include:

- gettimeofday() - Yes, its time can be changed, but does this matter for simulation?
- C++11 chrono library
- [Hack](http://stackoverflow.com/a/17112198/2548418) just for OS X

Are there are other constraints? Could one of the above suggestions work?

aswaterman, 2016-09-13:
I would favor not depending on the current time at all; it makes the simulator nondeterministic.  Could just use an affine function of the instructions-retired counter.

sbeamer, 2016-09-13:
Could it be the same as instructions-retired? Maybe just instructions-retired x 2?

aswaterman, 2016-09-13:
something like that would probably be ok

aswaterman, 2016-09-13:
(in spike, they are equal)

vapier, 2016-09-13:
`clock_gettime` is not Linux specific.  it's explicitly in POSIX:
  http://pubs.opengroup.org/onlinepubs/9699919799/functions/clock_getres.html

we can fallback to `CLOCK_MONOTONIC` if `CLOCK_BOOTTIME` is not available.  `getimeofday` is not appropriate as it's not monotonic -- it can be rewound.  any C++ lib is not appropriate as the GNU sim is C only, and doesn't rely on external C++ libs.

if none of those are available because the OS is crap, we can fallback to `csr.instret`.

aswaterman, 2016-09-13:
Trying all three is a reasonable approach, too... @vapier what do you think about the tradeoff between having an accurate time and having deterministic simulator runs?

palmer-dabbelt, 2017-02-08:
This is fixed, CSR_TIME now returns the cycle count.

closed 2017-02-08 by palmer-dabbelt