riscvarchive/riscv-binutils-gdb #24: Simulator does not build on ARM

maikmerten, 2016-10-04:
I'm trying to build the RISC-V toolchain on a Raspberry Pi 3 (which later on is supposed to assemble randomly generated RISC-V assembly for automated over-night tests of a RV32-core). The Raspberry Pi runs Raspian, which is basically Debian for ARMv6.

Building the simulator fails at sim-main.c, line 607:
```
#ifdef __GNUC__
  return ((__int128)a * b) >> 64;
#else
 // alternative implementation here
#endif
```

Turns out that checking for __GNUC__ is not sufficient to guarantee that __int128 is available:

https://gcc.gnu.org/onlinedocs/gcc/_005f_005fint128.html

> There is no support in GCC for expressing an integer constant of type __int128 for targets with long long integer less than 128 bits wide. 

According to http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.den0024a/ch08s02.html long long on ARM is 64 bits, not 128 bits, suggesting that __int128 is not available there.

The solution may be to ensure that the alternative code path is chosen. A clunky solution may be something like

`#if defined(__GNU__) && !defined(__ARM__)`

although a less platform-specific check may be possible and preferable.

kito-cheng, 2016-10-18:
@palmer-dabbelt I prefer remove the code for __int128 since it has portability issue and another potential bug https://github.com/kito-cheng/riscv-binutils-gdb/commit/7d44cc8472034c3b57221bdae72eae98b1df1834

palmer-dabbelt, 2016-10-19:
https://github.com/riscv/riscv-binutils-gdb/commit/6ceb705734e758daa3f2c761194f18136e112bb8

Thanks!

closed 2016-10-19 by palmer-dabbelt