riscvarchive/riscv-binutils-gdb #28: s_align doesn't handle 3 arguments

palmer-dabbelt, 2016-11-01:
This came as part of the binutils submission feedback

````
Hi Palmer,

  Just noticed this:

> diff --git a/gas/config/tc-riscv.c b/gas/config/tc-riscv.c
[...]
> +/* Pseudo-op table.  */
> +static const pseudo_typeS riscv_pseudo_table[] =
> +  {"p2align", s_align, 0},

This breaks one of the gas testsuite's tests (gas/testsuite/gas/all/align.s)
because it does not support the third argument to the normal
.p2align pseudo-op.  It also means that .p2align pseudo-op does
not behave as documented when the target is the riscv.

Please could you consider extending your s_align() function to
handle the extra argument.

Cheers
  Nick
````

aswaterman, 2016-11-01:
Should be addressed with 11e02552073f1d9e3f4894b855ddd1f631a5a9d7

palmer-dabbelt, 2016-11-01:
That's now

````
commit d8830018b884136ad8748a5097afd06a2ce26aef
Author: Andrew Waterman <waterman@cs.berkeley.edu>
Date:   Tue Nov 1 12:29:14 2016 -0700

    Don't define our own implementation of p2align
    
    Instead, define md_do_align to override handling of code alignment.
````

closed 2016-11-01 by palmer-dabbelt