riscvarchive/riscv-binutils-gdb #29: wcwidth() implicit declaration of function

ronhao, 2016-11-07:
Hi, when building I received the warning of implicit declaration of function of wcwidth() in

riscv-binutils-gdb/readline/mbutil.c

I googled it and found out that it is included wchar.h, but I didn't see #include<wchar.h> in mbutil.c. Do you think you missed it or are there any other explanations for this warning?

palmer-dabbelt, 2016-11-07:
This feels like an upstream issue (since it's in generic code), but it doesn't happen when I build the x86 version.  It's possible we just have some header that's smaller than the x86 version and therefor doesn't include <wchar.h>.  Can you paste the exact error message so I can make sure I'm not grepping incorrectly?

vapier, 2016-11-07:
yes, any issues in the readline/ subdir are an upstream problem.  you can use `--disable-readline --with-system-readline` to work around it (assuming you have readline dev packages installed in your system).  or just ignore it altogether :).  practically speaking, it should be harmless.

closed 2016-11-07 by vapier