riscvarchive/riscv-binutils-gdb #30: Fatal error: internal error: bad relocation #72

kito-cheng, 2016-11-10:
GCC: https://github.com/riscv/riscv-gcc/commit/2540fd6392506d3b8a0181db3ea72f5fdeca935e
Binutils: https://github.com/riscv/riscv-binutils-gdb/commit/2dfac042243e9bbf8cdb7fe2ffe6d662f868deb0

Arch: RV32* and RV64*
Option: -g -gdwarf-2

How to reproduce:
```
$ riscv64-unknown-linux-gnu-gcc pr65771.c -g -gdwarf-2 -c
/tmp/cctBkJjP.s: Assembler messages:
/tmp/cctBkJjP.s: Fatal error: internal error: bad relocation #72

```

Code for pr65771.c:
```
/* PR debug/65771 */
/* { dg-do link } */
/* { dg-require-effective-target tls_runtime } */
/* { dg-add-options tls } */

struct S { int s; int t; };
__thread struct S a[10];
int b;

int
main ()
{
  int c = a[b].t;
  (void) c;
  return 0;
}
```

aswaterman, 2016-11-12:
# 72 is BFD_RELOC_NONE.  Sounds like I messed something up : )

palmer-dabbelt, 2016-11-12:
I had this and fixed it by getting the right versions of all the tools.
riscv-gnu-toolchain is a bit behind because of all the upstream stuff in
flight, but I think if you bump binutils and gcc (and then drop GDB) it
should all build.  IIRC this is the result of shuffling around the
relocations so .option worked.

aswaterman, 2016-11-12:
I'm on master-everything, so I think it's a real bug. This one's probably an easy fix, at least (the last one took me 12 hours to make a 3-line fix :-/)

aswaterman, 2016-11-12:
By master I meant HEAD

kito-cheng, 2016-11-13:
Any idea?
```
(gdb) b tc-riscv.c:1629
Breakpoint 2 at 0x550fa6: file /home/kito/riscv-workspace/riscv-gnu-toolchain/riscv-binutils-gdb/gas/config/tc-riscv.c, line 1629.
(gdb) r
The program being debugged has been started already.
Start it from the beginning? (y or n) y
Starting program: /home/kito/riscv-workspace/riscv-gnu-toolchain/install/bin/riscv64-unknown-linux-gnu-as test.s
warning: the debug information found in "/lib64/ld-2.23.so" does not match "/lib64/ld-linux-x86-64.so.2" (CRC mismatch).

[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".

Breakpoint 2, riscv_ip (str=0x6310000009e0 "add", ip=0x7fffffffe4e0, imm_expr=0x7fffffffe4a0, imm_reloc=0x7fffffffe460)
    at /home/kito/riscv-workspace/riscv-gnu-toolchain/riscv-binutils-gdb/gas/config/tc-riscv.c:1629
1629		      p = riscv_opts.relax ? percent_op_rtype_relax : percent_op_rtype;
(gdb) p riscv_opts.relax
$10 = 0
(gdb) p percent_op_rtype_relax
$11 = {{str = 0x769500 "%tprel_add", reloc = BFD_RELOC_RISCV_TPREL_ADD}, {str = 0x0, reloc = _dummy_first_bfd_reloc_code_real}}
(gdb) p percent_op_rtype
$12 = {{str = 0x769500 "%tprel_add", reloc = BFD_RELOC_NONE}, {str = 0x0, reloc = _dummy_first_bfd_reloc_code_real}}
(gdb) 

```

aswaterman, 2016-11-13:
I'm working on it now.

commit 73354bd1 (2016-11-13): Use BFD_RELOC_UNUSED, not BFD_RELOC_NONE, to indicate no reloc

Resolves #30.
closed 2016-11-13 by aswaterman