riscvarchive/riscv-binutils-gdb #31: Dwarf Error: mangled line number section.

kito-cheng, 2016-11-10:
GCC: riscv/riscv-gcc@2540fd6
Binutils: 2dfac04
Glibc: https://github.com/riscv/riscv-glibc/commit/7ba28f3cc8990505a1a9f01b93a27d8d79527bc2

Arch: RV32IMA and RV32G

How to reproduce:
```
$ riscv32-unknown-linux-gnu-gcc anon5.C  -gdwarf-2
/home/kito/riscv-workspace/rv32ima/lib/gcc/riscv32-unknown-linux-gnu/6.1.0/../../../../riscv32-unknown-linux-gnu/bin/ld: Dwarf Error: mangled line number section.
/tmp/ccvn2QJe.o: In function `.LCFI1':
anon5.C:(.text+0xc): undefined reference to `(anonymous namespace)::c::t'
anon5.C:(.text+0x10): undefined reference to `(anonymous namespace)::c::t'
collect2: error: ld returned 1 exit status
```
Note: undefined reference is expected output.

Code for anon5.C:
```
// PR c++/34094
// { dg-do link { target { ! { *-*-darwin* *-*-hpux* *-*-solaris2.* } } } }
// { dg-options "-gdwarf-2" }
// Ignore additional message on powerpc-ibm-aix
// { dg-prune-output "obtain more information" } */
// Ignore additional messages on Linux/x86 with PIE
// { dg-prune-output "Bad value" } */

namespace {
  struct c
  {
    static const bool t = 0;
  };
}

const bool &f()
{
  return c::t;	// { dg-message "\[Uu\]ndefined" "undefined" { target *-*-* } 0 }
		// Some targets report the error for the previous line, others
		// don't give line number inforamtion for it, so use line 0.
}

int main(void)
{
  return 0;
}
```

kito-cheng, 2016-11-13:
This bug seem gone with https://github.com/riscv/riscv-binutils-gdb/commit/49d8e1089b52102827113a9deb2a5627cf0b46eb, so I close it

closed 2016-11-13 by kito-cheng

aswaterman, 2016-11-13:
Nice.