riscvarchive/riscv-binutils-gdb #32: Build failed on macOS because of missing library "rt"

pbing, 2016-11-26:
```
gcc -DHAVE_CONFIG_H -DWITH_DEFAULT_MODEL='"RV64G"' -DWITH_ALIGNMENT=NONSTRICT_ALIGNMENT -DWITH_TARGET_WORD_BITSIZE=64 -DWITH_TARGET_BYTE_ORDER=BFD_ENDIAN_LITTLE   -DDEFAULT_INLINE=0   -Wall -Wdeclaration-after-statement -Wpointer-arith -W
          nrun.o libsim.a ../../bfd/libbfd.a ../../opcodes/libopcodes.a ../../intl/libintl.a -liconv ../../libiberty/libiberty.a -L../../zlib -lz -lrt -ldl
```

```
ld: library not found for -lrt
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[4]: *** [run] Error 1
make[3]: *** [all] Error 1
make[2]: *** [all-sim] Error 2
make[1]: *** [all] Error 2
make: *** [stamps/build-binutils-newlib] Error 2
```

macOS Sierra 10.12.1
Xcode 8.1
riscv-gnu-toolchain commit 87ec00322ed6caa207747ce00ed38ca15995ec63

aswaterman, 2016-11-26:
Are you using homebrew? There's probably a package for librt.

pbing, 2016-11-26:
[Seems](https://github.com/Homebrew/legacy-homebrew/commit/83235bf1b6ce8f99073546ecf18cbd84ba72b222) that macOS does not use library 'rt' for clock_time().

pbing, 2017-01-21:
Works now with 2917a9f501cf2385b5e02fffeeb36313b3d6a1d0.

closed 2017-01-21 by pbing