riscvarchive/riscv-binutils-gdb #33: The readme for gdb needs to be made specific for gdb

rminnich, 2016-12-01:
It's probably just me but I'm hopelessly confused by the build and install process here. What environment variables if any do I need to set to make sure the right binary gets built, and where it gets installed? What branch should I use?

palmer-dabbelt, 2016-12-01:
The riscv-binutils-gdb repository is meant to be as close to upstream as possible, so any README you see in here is from upstream.  If you're looking for build instructions for a RISC-V userspace, the best documentation is the Makefile.in inside riscv-gnu-toolchain <https://github.com/riscv/riscv-gnu-toolchain/blob/master/Makefile.in#L122>.  That doesn't pass "--disable-gdb" to the binutils configure invocation so it builds a RISC-V GDB (which is the default behavior) as part of building the rest of binutils.

You don't need to set any environment variables when building riscv-gnu-toolchain, but you should pass --prefix to configure as you probably don't want to install to /usr/local.  The RISCV environment variable is only used for rocket-chip and riscv-tools.

We're currently in the middle of upstreaming so the branches are a bit of a mess.  The easiest way to get compatible versions of everything is to use what's tagged in rocket-chip (via riscv-tools and then riscv-gnu-toolchain), but if you just care about software then using what's in riscv-gnu-toolchain might be a bit more up to date.

palmer-dabbelt, 2017-02-13:
I'm closing this, as it's the README from upstream.

closed 2017-02-13 by palmer-dabbelt