riscvarchive/riscv-binutils-gdb #34: gdb: unimplemented breakpoint_kind_from_pc method

a0u, 2016-12-06:
gdb built from 2e8a797d9eaef6ad5bd072899b25a1e6e6efce21 aborts with the following error:
```
/usr/local/src/riscv-gnu-toolchain/riscv-binutils-gdb/gdb/gdbarch.c:701: internal-error: verify_gdbarch: the following are invalid ...
        breakpoint_kind_from_pc
A problem internal to GDB has been detected,
further debugging may prove unreliable.
Quit this debugging session? (y or n)
```

It appears that `riscv-tdep.c` is missing the gdbarch method `breakpoint_kind_from_pc`, which was added by upstream commit cd6c3b4ffc4ea6f56f12581419e2eed885441beb.

palmer-dabbelt, 2016-12-08:
@a0u I don't have anything setup to test GDB, can you look?  I'm going to close this bug so we can move this to the PR.

closed 2016-12-08 by palmer-dabbelt

timsifive, 2016-12-08:
You can test gdb by running 'make clean; make all' in riscv-tests/debug

timsifive, 2016-12-08:
I have done that. The bug exists in `riscv-next`. It's gone in `breakpoint_from_pc`.

PR #37 (2016-12-08): Implement breakpoint_kind_from_pc, sw_breakpoint_from_kind
This should fix #34
commit 727f2917 (2016-12-08): Implement breakpoint_kind_from_pc, sw_breakpoint_from_kind

This should fix #34
commit 79834996 (2016-12-16): Implement breakpoint_kind_from_pc, sw_breakpoint_from_kind

This should fix #34
commit 0b6d7568 (2016-12-18): Implement breakpoint_kind_from_pc, sw_breakpoint_from_kind

This should fix #34