riscvarchive/riscv-binutils-gdb #35: Compilation fails on Mac OSX due to 'clock_gettime'

mwachs5, 2016-12-06:
Not sure this is the same/related to #32. I also cannot compile on MAC OSX because of clock_gettime doesn't seem to be implemented on MAC:

```
 /Users/megan/freedom-e-sdk/riscv-gnu-toolchain/riscv-binutils-gdb/sim/riscv/sim-main.c:96:6: warning: 
      implicit declaration of function 'clock_gettime' is invalid in C99
      [-Wimplicit-function-declaration]
        if (clock_gettime (CLOCK_MONOTONIC, &ts) == 0)
            ^
/Users/megan/freedom-e-sdk/riscv-gnu-toolchain/riscv-binutils-gdb/sim/riscv/sim-main.c:96:21: error: 
      use of undeclared identifier 'CLOCK_MONOTONIC'
        if (clock_gettime (CLOCK_MONOTONIC, &ts) == 0)
```

palmer-dabbelt, 2016-12-06:
The latest version (on riscv-next) of the RISC-V port shouldn't have clock_gettime() in sim.  This is part of the upstream cleanup process, which has gotten a bit bogged down because we have to deal with some ABI issues.

You can just build with "--disable-sim", since you're probably not using the GNU sim.

palmer-dabbelt, 2017-02-13:
I'm closing this, I believe it's fixed.

closed 2017-02-13 by palmer-dabbelt