riscvarchive/riscv-binutils-gdb #36: objdump disassembly gives wrong instruction mnemonic csrrc instead of csrrci

makigumo, 2016-12-07:
Hello, I think there is a small error in the disassembly.

The following instruction `1000f9f3` (in big endian) is disassembled to `csrrc   s3,sstatus,1` but should actually be `csrrci   s3,sstatus,1` as seems obvious by the immediate operand `1`. This is further backed upd by the spec 2.1 where `csrrci` is given for the `funct3` of `0b111`.

I'm using the bottled package (e.g. prebuilt) from https://github.com/riscv/homebrew-riscv

Thanks.

aswaterman, 2016-12-07:
Hi.  csrrc can be used as a pseudo-instruction for csrrci.  However, the
disassembler should display it like you proposed.  I will fix it.

In general, the disassembler might choose a more idiomatic form, unless you
pass the flag -Mno-aliases.  For example, it displays "ret" instead of
"jalr zero, 0(ra)" by default.

makigumo, 2016-12-07:
Thank you for your clarification. I'was only aware of the pseudo-instructions mentioned in the spec. Is there a different document containing those extra pseudo-instructions?

aswaterman, 2016-12-07:
The assembly programmer's manual is incomplete, but as I mentioned, if you notice a pseudo-instruction you don't recognize, you can disassemble with -Mno-aliases to identify it.

commit 461c0c6f (2016-12-07): Fix disassembly of CSR instructions under -Mno-aliases

Resolves #36.
closed 2016-12-07 by aswaterman