riscvarchive/riscv-binutils-gdb #38: Instruction-format pseudos

sorear, 2016-12-09:
See riscv/riscv-gnu-toolchain#190

```
r_format opcode, funct3, func7, rd, rs1, rs2
i_format opcode, funct3, rd, rs1, immed
s_format opcode, funct3, rs1, rs2, immed
sb_format opcode, funct3, rs1, rs2, immed (or label?)
u_format opcode, rd, immed
uj_format opcode, rd, immed (or label?)
r4_format opcode, funct3, funct2, rd, rs1, rs2, rs3
cr_format op, funct4, rd/rs1, rs2
... C-space is pretty full so it may not make sense
```

`.word` doesn't support register names, so it's harder to use with gcc inline asm.

aswaterman, 2016-12-09:
I agree there's little reason to support the RVC formats.  Might even just start with R and I and see what sticks.

sorear, 2017-07-26:
https://sourceware.org/binutils/docs-2.29/as/s390-Directives.html `.insn` seems like a good model to follow

aswaterman, 2017-07-26:
Stuffing it all in ```.insn``` will avoid namespace pollution, and can provide slightly better error messages.

zhenbohu, 2017-09-03:
Dear Experts

I recently just found my old RoCC prototype is not working anymore because of the toolchain have dropped the "Custom" instruction supporting. It was shocking to me and I feel this is really a upset. And by google and reading the issue https://github.com/riscv/riscv-gnu-toolchain/issues/190 I got to know it was decided  to switch to the scheme filed by this issue. But if I understand it correctly until now this scheme is not yet really implemented in toolchain now?

That means, right  now the GCC toolchain have no way to add the user-extension instructions (no custom supported, and no this new scheme implemented)? 

Extension is the biggest weapon for RISC-V, when will we re-arm it? is  there any plan or schedule for it?

Thanks
Bob

aswaterman, 2017-09-04:
If someone has the time to prototype @sorear's proposal in GAS (something like ```.insn r, opcode, funct3, func7, rd, rs1, rs2```), we would certainly review it and consider it for inclusion.  @palmer-dabbelt and I just don't have time to do it ourselves right now.  @kito-cheng maybe Andes also wants this feature, so you can get some time to do it?

kito-cheng, 2017-09-05:
Hummm, ya, it's useful to us, but it's not must to have item to us since we'll implement by `riscv-opcode` + `binutils` directly :P
Anyway, I'll put it into my personal TODO list.

palmer-dabbelt, 2017-09-05:
@zhenbohu You can use the macro-based scheme, but not the assembler-based one.  See here

https://github.com/riscv/riscv-gnu-toolchain/issues/190#issuecomment-264083212

zhenbohu, 2017-09-08:
@palmer-dabbelt

Hi, Palmer

I have already read this before, but that scheme is not a good enough scheme, because it may need additional "mv" instructions to cause performance degradation. So the native support for extension instruction would be greatly better.

Thanks
Bob

palmer-dabbelt, 2017-09-08:
Ya, I understand that -- it's just that this isn't a priority for us right now.  If you're interested in hacking on the toolchain I can give you some pointers, it should be an easy thing to implement.  If not, then hopefully we'll get some time to do this right but there's no timeline right now.  Sorry!

zhenbohu, 2017-09-10:
@palmer-dabbelt
 
Thanks very much for your kind. Although I am very interested on it, but I am a purely hardware guy, not sure if I can really handle it. Maybe I will still wait for your official support on it in toolchain.

kito-cheng, 2018-03-16:
.insn directive is merged into upstream now :)

https://sourceware.org/git/gitweb.cgi?p=binutils-gdb.git;a=commit;h=0e35537d754f1c687850d1caccb2d78d2e418391

aswaterman, 2018-03-16:
Nice!

closed 2018-03-16 by aswaterman