riscvarchive/riscv-binutils-gdb #39: Missing hideleg/hedeleg and sideleg/sedeleg CSRs

michaeljclark, 2016-12-14:
Hi,

I’ve implemented the mideleg/medeleg, hideleg/hedeleg and sideleg/sedeleg interrupt cascade mechanism.

The implementation I am working on has H-mode thus I've set mideleg/medeleg and hideleg/hedeleg to cascade interrupts to S-mode.

I’m getting an error with the toolchain (built from master) and it appears the hideleg/hedeleg (and likely sideleg/sedeleg) are not

src/rom/boot-pgtable.c: Assembler messages:
src/rom/boot-pgtable.c:131: Error: Instruction csrw requires absolute expression
src/rom/boot-pgtable.c:133: Error: Instruction csrw requires absolute expression
make[1]: *** [build/riscv64-unknown-elf/obj/boot-pgtable.o] Error 1

$ nl -b a src/rom/boot-pgtable.c | egrep '(131|133)'
   131	  write_csr(hideleg, interrupts);
   133	  write_csr(hedeleg, exceptions);

Michael

michaeljclark, 2016-12-14:
I think something like the following needs to be added to `riscv-binutils-gdb/include/opcode/riscv-opc.h`. I will try locally.

```
#define CSR_HEDELEG 0x202
#define CSR_HIDELEG 0x203
#define CSR_SEDELEG 0x102
#define CSR_SIDELEG 0x103
DECLARE_CSR(hedeleg, CSR_HEDELEG)
DECLARE_CSR(hideleg, CSR_HIDELEG)
DECLARE_CSR(sedeleg, CSR_SEDELEG)
DECLARE_CSR(sideleg, CSR_SIDELEG)
```

palmer-dabbelt, 2016-12-15:
The CSR instructions take indicies as arguments in addition to names, so you can work around this without a toolchain change.

palmer-dabbelt, 2017-03-06:
@aswaterman Are this going into the next supervisor spec, or should I close this bug?  I see only medeleg/mideleg in parse-opcodes.

aswaterman, 2017-03-07:
Let's close this for now.  We are punting both the hypervisor mode and user-level interrupts to the version after next, so neither set of registers will be in the coming version.

closed 2017-03-07 by palmer-dabbelt