riscvarchive/riscv-binutils-gdb #50: _gp

palmer-dabbelt, 2017-01-02:


sorear, 2017-01-02:
What about it?

palmer-dabbelt, 2017-01-03:
We copied this symbol name from MIPS, but Andrew realized it's probably not a good idea as it's not in any protected namespace.  While it's probably OK for most software because of the MIPS heritage, we figured we might as well change it while we still can.

The plan was to go figure out what other people are doing, rule out all the bad names (like _gp), and then ask you if you had any opinions.  You're welcome to do a bit more work if you want... :).

aswaterman, 2017-01-03:
As we discussed earlier, it is probably benign, since we are not the only ISA that uses this name.  But if we do change it, it should be to something with two underscores.

vapier, 2017-01-03:
a simple switch like `__gp` would be good to do early.  if possible, using a symbol name that isn't valid from the C ABI might be worth investigating.  like `__gp$` or `__global_pointer$`.

palmer-dabbelt, 2017-02-08:
Fixed in #56, hopefully it's OK for upstream.

closed 2017-02-08 by palmer-dabbelt