riscvarchive/riscv-binutils-gdb #53: Cannot build binutils on MSYS2/Windows 7- ld.info target fails

cr1901, 2017-01-21:
When attempting to build riscv-binutils from riscv-gnu-toolchain, I get the following error when building ld.info:

```
if (makeinfo --split-size=5000000 --split-size=5000000 --version) >/dev/null 2>&1; then \
  for f in ld.info ld.info-[0-9] ld.info-[0-9][0-9] ld.i[0-9] ld.i[0-9][0-9]; do \
    if test -f $f; then mv $f $backupdir; restore=mv; else :; fi; \
  done; \
else :; fi && \
if makeinfo --split-size=5000000 --split-size=5000000 -I /home/William/src/riscv-gnu-toolchain/riscv-binutils-gdb/ld -I /home/William/src/riscv-gnu-toolchain/riscv-binutils-gdb/ld/../bfd/doc -I ../bfd/doc -I /home/William/src/riscv-gnu-toolchain/riscv-binutils-gdb/ld/../libiberty  -I /home/William/src/riscv-gnu-toolchain/riscv-binutils-gdb/ld \
 -o ld.info `test -f 'ld.texinfo' || echo '/home/William/src/riscv-gnu-toolchain/riscv-binutils-gdb/ld/'`ld.texinfo; \
then \
  rc=0; \
else \
  rc=$?; \
  $restore $backupdir/* `echo "./ld.info" | sed 's|[^/]*$||'`; \
fi; \
rm -rf $backupdir; exit $rc
/home/William/src/riscv-gnu-toolchain/riscv-binutils-gdb/ld/ld.texinfo:358: table requires an argument: the formatter for @item
/home/William/src/riscv-gnu-toolchain/riscv-binutils-gdb/ld/ld.texinfo:2428: table requires an argument: the formatter for @item
/home/William/src/riscv-gnu-toolchain/riscv-binutils-gdb/ld/ld.texinfo:2908: table requires an argument: the formatter for @item
/home/William/src/riscv-gnu-toolchain/riscv-binutils-gdb/ld/ld.texinfo:2940: table requires an argument: the formatter for @item
/home/William/src/riscv-gnu-toolchain/riscv-binutils-gdb/ld/ld.texinfo:2968: table requires an argument: the formatter for @item
/home/William/src/riscv-gnu-toolchain/riscv-binutils-gdb/ld/ld.texinfo:2990: table requires an argument: the formatter for @item
'home/William/src/riscv-gnu-toolchain/riscv-binutils-gdb/ld/ld.texinfo:4618: unknown command `
'home/William/src/riscv-gnu-toolchain/riscv-binutils-gdb/ld/ld.texinfo:4622: unknown command `
Makefile:1572: recipe for target 'ld.info' failed
make[4]: *** [ld.info] Error 1
make[4]: Leaving directory '/home/William/src/riscv-gnu-toolchain/build-binutils-newlib/ld'
Makefile:1757: recipe for target 'all-recursive' failed
make[3]: *** [all-recursive] Error 1
make[3]: Leaving directory '/home/William/src/riscv-gnu-toolchain/build-binutils-newlib/ld'
Makefile:989: recipe for target 'all' failed
make[2]: *** [all] Error 2
make[2]: Leaving directory '/home/William/src/riscv-gnu-toolchain/build-binutils-newlib/ld'
Makefile:7181: recipe for target 'all-ld' failed
make[1]: *** [all-ld] Error 2
make[1]: Leaving directory '/home/William/src/riscv-gnu-toolchain/build-binutils-newlib'
Makefile:857: recipe for target 'all' failed
make: *** [all] Error 2
```

For reference, I built binutils 2.27 from source recently on Windows (1-8-2017) to report a bug, and there were no issues compiling. In fact, according to my still-present build-tree, `ld.info` wasn't built at all, although there is nothing to suggest in [the MinGW patches](https://github.com/Alexpux/MINGW-packages/tree/master/mingw-w64-binutils) that building info files is disabled.

`touch ld/ld.info` in the build tree to skip building works fine on Windows, but I wonder what causes the target to be build at all compared to a "vanilla" MinGW binutils build.

aswaterman, 2017-01-21:
This strikes me as a generic binutils problem, unrelated to RISC-V.  Can you try building the binutils trunk from source?  https://sourceware.org/git/gitweb.cgi?p=binutils-gdb.git

It's possible that something relevant in the binutils mainline has changed since 2.27.

vapier, 2017-01-21:
agreed -- inability to generate info pages should be taken up with the environment or upstream.  RISC-V isn't doing anything special here.

cr1901, 2017-01-21:
Yes, apologies, just confirmed the issue happens w/ a vanilla binutils trunk. Which means the error was introduced recently...

cr1901, 2017-01-21:
This is a bug with texinfo. Sorry for the inconvenience :P.

closed 2017-01-21 by cr1901

aswaterman, 2017-01-21:
Thanks for checking, and good luck getting it resolved.