riscvarchive/riscv-binutils-gdb #54: gdb can't distinguish 32- from 64-bit targets

timsifive, 2017-01-26:
If you start a 64-bit gdb, and immediately connect it to a 32-bit target, gdb will think the target is 32 bits. Register values won't be interpreted correctly, and generally the setup is unusable. If you give gdb an ELF file, it will find XLEN from there. You can also explicitly force gdb to 32-bit using `set arch riscv:rv32`.

To solve this nicely, the server gdb connects to (eg. OpenOCD) would have to return a different target description for RV32 versus RV64 cores. https://sourceware.org/ml/gdb/2016-10/msg00015.html describes roughly what work is required to accomplish this.

TM1234, 2017-11-02:
I presume that there is a typo in the original post and...

> If you start a 64-bit gdb, and immediately connect it to a 32-bit target, gdb will think the target is 32 bits. 

Should read...

> If you start a 64-bit gdb, and immediately connect it to a 32-bit target, gdb will think the target is 64 bits. 

Is that correct?

timsifive, 2017-11-03:
@TM1234: Yes. I fixed it in the original comment.

TM1234, 2017-11-03:
Thanks Tim. :-)

timsifive, 2017-12-20:
Confirmed that this is fixed with gdb/openocd from their respective xml_registers branches (which have merged to their mainlines).

closed 2017-12-20 by timsifive

TM1234, 2017-12-20:
Does this mean that it's no longer necessary to explicitly pass set arch riscv:rv32 when connecting to a 32 bit target?

timsifive, 2017-12-21:
Yes.