riscvarchive/riscv-binutils-gdb #55: How does GDB Handle non-existent MISA?

mwachs5, 2017-02-06:
The code for determining whether a core implements C extension uses MISA, which is optional. How does/should GDB handle registers that don't exist?

The code 
https://github.com/riscv/riscv-binutils-gdb/blob/d575c2c2ba936ad70cc87038e3332b9dac5a3c16/gdb/riscv-tdep.c#L166

 "tries to read MISA", but it doesn't seem to check for any error status that might be returned by the target if MISA doesn't exist. 

Should there be a canonical "not implemented" value returned if MISA is not implemented (as detected at a lower level)? Or should this code be checking the response from the target when it tries to read MISA?

aswaterman, 2017-02-06:
The MISA CSR must always exist, but if not implemented reads as zero.

That said, I think it's OK to restrict debugging support to systems that implement MISA.

mwachs5, 2017-02-06:
Ah, OK. I misunderstood that MISA should always be accessible even if not implemented.

closed 2017-02-06 by mwachs5