riscvarchive/riscv-binutils-gdb #59: Enhance assembler/disassembler output for lui/c.lui

kito-cheng, 2017-02-16:
Arch: RV*C

**Test case 1 for disassembler:**
```
$ cat t1.s
x:
  li s5, -4096
$ riscv64-unknown-elf-gcc t1.s -c
$ riscv64-unknown-elf-objdump -d t1.o
0000000000000000 <x>:
   0:	7afd                	lui	s5,0xfffff
```
Expected result:
```
0000000000000000 <x>:
   0:	7afd                	lui	s5,-4096
```

**Test case 2 for disassembler:**

```
$ cat t4.s 
x:
  li s5, -4096
$ riscv64-unknown-elf-gcc t4.s -c -march=rv64ima
$ riscv64-unknown-elf-objdump -d t4.o 
0000000000000000 <x>:
   0:	fffffab7          	lui	s5,0xfffff
```
Expected result:
```
0000000000000000 <x>:
   0:	fffffab7          	lui	s5,0xfffff
```

**Test case 3 for assembler:**
```
$ cat t2.s 
x:
  c.lui s5, -4096
$ riscv64-unknown-elf-gcc t2.s -c
t2.s: Assembler messages:
t2.s:2: Error: illegal operands `c.lui s5,-4096'
```
Expected result:
Accept the input.

**Test case 4 for assembler:**
```
$ cat t3.s
x:
  lui s5, -4096
$ riscv64-unknown-elf-gcc t3.s -c -march=rv64ima
t3.s: Assembler messages:
t3.s:2: Error: lui expression not in range 0..1048575
```
Expected result:
Accept the input.

aswaterman, 2017-02-16:
```lui``` takes the immediate right-shifted by 12 and unsigned (i.e. 0x0 - 0xfffff), so -4096 would be 0xfffff.  I fear that changing this will cause problems.  In your proposal, what range of inputs would lui accept?

Seems better to rely on ```li``` if you want to use constants like -4096; this is what GCC does.

kito-cheng, 2017-02-17:
Humm, I expect it `lui` can accept two kind of input, shifted immediate and signed integer.
For shifted immediate, accept 0xfffff000 instead of 0xfffff and output 0xfffff000 in disassembler.
For signed integer, accept all encode-able number.

And for disassembler, naybe we can add both kind of output?
``
0000000000000000 <x>:
   0:	fffffab7          	lui	s5,0xfffff000 # -4096
```

aswaterman, 2017-02-17:
If we changed it to accept 0xfffff000 instead of 0xfffff, this would make sense, but I really don't want to make a backwards-incompatible change like this.  It won't break gcc, but it will break hand-written assembly.

I think in the style guide, we just tell people to use li, not lui.  We can make it a requirement that li uses lui if the result is in lui range, so that it is predictable.

kito-cheng, 2017-02-17:
OK, I agree it's will break backwards-compatible, and the let the disassembler output keep the same to prevent inconsistent or any confuse.

closed 2017-02-17 by kito-cheng