riscvarchive/riscv-binutils-gdb #60: GDB tries to access FPRs without F extension

mwachs5, 2017-02-17:
When GDB wants to make a function call while debugging, it saves registers and restores them afterwards. Currently it seems to save FPRs even if there is no F extension. Should it be checking for F before doing this? Or, what is the expected behavior here?

This is currently working on real hardware without F because OpenOCD wasn't reporting errors all the way to GDB when it accessed non-existent registers (GDB just treated whatever junk OpenOCD returned as the value, and since it didn't matter when you restored, it was fine). 

We've have a branch of OpenOCD which report the errors now, so should GDB check for the presence of F before saving the FPRs (and so on as more extensions show up...), or otherwise handle errors returned for nonexistent registers? Or is returning an error to GDB for a nonexistent register the wrong thing to do?

timsifive, 2017-07-19:
I'm working on this, but can't figure out how to self-assign.

timsifive, 2017-07-21:
Fixed by PR #95.

closed 2017-07-21 by timsifive