riscvarchive/riscv-binutils-gdb #64: objdump: srli s0,s0,xx does not roundtrip

sorear, 2017-03-13:
c.srli s0,xx is generated instead.  Other registers and compressed instructions appear to be fine.

palmer-dabbelt, 2017-03-13:
I'm not quite sure what you're asking, but I do see some odd behavior

````
palmer gamma02 ~ $ cat shift.S 
.global main
main:
	addi s0, s0, 1
	srli s0, s0, 1
	srli s1, s1, 2
	ret
palmer gamma02 ~ $ riscv64-unknown-linux-gnu-gcc -march=rv64gc shift.S -o shift.o -c
palmer gamma02 ~ $ riscv64-unknown-linux-gnu-objdump -d shift.o

shift.o:     file format elf64-littleriscv

Disassembly of section .text:

0000000000000000 <main>:
   0:	0405                	addi	s0,s0,1
   2:	00145413          	srli	s0,s0,0x1
   6:	8089                	srli	s1,s1,0x2
   8:	8082                	ret
````

This does look odd to me, though -- I'd expect s0 to get compressed just like the s1.  @aswaterman Am I  crazy, or should these instructions all compress?

sorear, 2017-03-13:
Sorry what I meant was this in disassembly output:

```
   14976:       0084f433                and     s0,s1,s0
   1497a:       1402                    slli    s0,s0,0x20
   1497c:       9001                    c.srli  s0,0x20
   1497e:       e419                    bnez    s0,1498c <runtime.(*hmap).incrnoverflow+0x5c>
```

I would have expected it to say `srli s0,s0,0x20`

aswaterman, 2017-03-13:
@palmer-dabbelt  https://github.com/riscv/riscv-binutils-gdb/commit/223c9d1fff425a7424fa450770e4cf4db822bf05 fixes one or both of these problems? I think it was a copy-paste screwup.

palmer-dabbelt, 2017-03-13:
Ya, I think that's the fix.  I'm testing it along with some other stuff.

jim-wilson, 2017-11-30:
I confirmed that the commit https://github.com/riscv/riscv-binutils-gdb/commit/223c9d1fff425a7424fa450770e4cf4db822bf05 fixes the problem.  So closing.

closed 2017-11-30 by jim-wilson