riscvarchive/riscv-binutils-gdb #69: riscv-binutils-gdb won't build on macos

mwachs5, 2017-04-10:
```
riscv-gnu-toolchain/riscv-binutils-gdb/gdb/common/gdb_unlinker.h:35:35: error: '__nonnull__' attribute is invalid for the implicit this argument
  unlinker (const char *filename) ATTRIBUTE_NONNULL (1)
```

aswaterman, 2017-04-10:
Looks like that's in code that's out of control... which gcc version are you using?  (or clang. for that matter)

mwachs5, 2017-04-10:
```
>> gcc --version
Configured with: --prefix=/Library/Developer/CommandLineTools/usr --with-gxx-include-dir=/usr/include/c++/4.2.1
Apple LLVM version 8.0.0 (clang-800.0.42.1)
Target: x86_64-apple-darwin15.6.0
Thread model: posix
InstalledDir: /Library/Developer/CommandLineTools/usr/bin****
```

aswaterman, 2017-04-10:
Seems like this is fixed upstream: http://patchwork.sourceware.org/patch/19480/

Can you manually apply that fix and see if it works?  If so, maybe we'll have to backport the patch to this branch... what do you think @palmer-dabbelt

palmer-dabbelt, 2017-04-10:
If it works then it'll be easy to maintain.

palmer-dabbelt, 2017-05-20:
This was at least building for the SDK release, so I think this has been fixed.

closed 2017-05-20 by palmer-dabbelt