riscvarchive/riscv-binutils-gdb #76: .eh_frame_hdr refers to overlapping FDEs when linking libc

palmer-dabbelt, 2017-05-08:
This is a regression from riscv-binutils-2.28 (3f21b5c9675db61ef5462442b6a068d4a3da8aaf) to riscv-next (9fc62d500cf36bb6c9d2360dfcd9397ddd417e4d).

````
riscv32-unknown-linux-gnu-gcc    -shared -static-libgcc -Wl,-O1  -Wl,-z,defs -Wl,-dynamic-linker=/lib/ld-linux-riscv32-ilp32d.so.1  -B/scratch/palmer/work/upstream/riscv-gnu-toolchain/build/build-glibc-linux-32/csu/  -Wl,--version-script=/scratch/palmer/work/upstream/riscv-gnu-toolchain/build/build-glibc-linux-32/libc.map -Wl,-soname=libc.so.6 -Wl,-z,combreloc -Wl,-z,relro -Wl,--hash-style=both  -nostdlib -nostartfiles -e __libc_main -L/scratch/palmer/work/upstream/riscv-gnu-toolchain/build/build-glibc-linux-32 -L/scratch/palmer/work/upstream/riscv-gnu-toolchain/build/build-glibc-linux-32/math -L/scratch/palmer/work/upstream/riscv-gnu-toolchain/build/build-glibc-linux-32/elf -L/scratch/palmer/work/upstream/riscv-gnu-toolchain/build/build-glibc-linux-32/dlfcn -L/scratch/palmer/work/upstream/riscv-gnu-toolchain/build/build-glibc-linux-32/nss -L/scratch/palmer/work/upstream/riscv-gnu-toolchain/build/build-glibc-linux-32/nis -L/scratch/palmer/work/upstream/riscv-gnu-toolchain/build/build-glibc-linux-32/rt -L/scratch/palmer/work/upstream/riscv-gnu-toolchain/build/build-glibc-linux-32/resolv -L/scratch/palmer/work/upstream/riscv-gnu-toolchain/build/build-glibc-linux-32/crypt -L/scratch/palmer/work/upstream/riscv-gnu-toolchain/build/build-glibc-linux-32/mathvec -L/scratch/palmer/work/upstream/riscv-gnu-toolchain/build/build-glibc-linux-32/support -L/scratch/palmer/work/upstream/riscv-gnu-toolchain/build/build-glibc-linux-32/nptl -Wl,-rpath-link=/scratch/palmer/work/upstream/riscv-gnu-toolchain/build/build-glibc-linux-32:/scratch/palmer/work/upstream/riscv-gnu-toolchain/build/build-glibc-linux-32/math:/scratch/palmer/work/upstream/riscv-gnu-toolchain/build/build-glibc-linux-32/elf:/scratch/palmer/work/upstream/riscv-gnu-toolchain/build/build-glibc-linux-32/dlfcn:/scratch/palmer/work/upstream/riscv-gnu-toolchain/build/build-glibc-linux-32/nss:/scratch/palmer/work/upstream/riscv-gnu-toolchain/build/build-glibc-linux-32/nis:/scratch/palmer/work/upstream/riscv-gnu-toolchain/build/build-glibc-linux-32/rt:/scratch/palmer/work/upstream/riscv-gnu-toolchain/build/build-glibc-linux-32/resolv:/scratch/palmer/work/upstream/riscv-gnu-toolchain/build/build-glibc-linux-32/crypt:/scratch/palmer/work/upstream/riscv-gnu-toolchain/build/build-glibc-linux-32/mathvec:/scratch/palmer/work/upstream/riscv-gnu-toolchain/build/build-glibc-linux-32/support:/scratch/palmer/work/upstream/riscv-gnu-toolchain/build/build-glibc-linux-32/nptl -o /scratch/palmer/work/upstream/riscv-gnu-toolchain/build/build-glibc-linux-32/libc.so -T /scratch/palmer/work/upstream/riscv-gnu-toolchain/build/build-glibc-linux-32/shlib.lds /scratch/palmer/work/upstream/riscv-gnu-toolchain/build/build-glibc-linux-32/csu/abi-note.o /scratch/palmer/work/upstream/riscv-gnu-toolchain/build/build-glibc-linux-32/elf/soinit.os /scratch/palmer/work/upstream/riscv-gnu-toolchain/build/build-glibc-linux-32/libc_pic.os /scratch/palmer/work/upstream/riscv-gnu-toolchain/build/build-glibc-linux-32/elf/sofini.os /scratch/palmer/work/upstream/riscv-gnu-toolchain/build/build-glibc-linux-32/elf/interp.os /scratch/palmer/work/upstream/riscv-gnu-toolchain/build/build-glibc-linux-32/elf/ld.so -lgcc
/scratch/palmer/work/upstream/riscv-gnu-toolchain/build/install/bin/../lib/gcc/riscv32-unknown-linux-gnu/8.0.0/../../../../riscv32-unknown-linux-gnu/bin/ld: .eh_frame_hdr refers to overlapping FDEs.
/scratch/palmer/work/upstream/riscv-gnu-toolchain/build/install/bin/../lib/gcc/riscv32-unknown-linux-gnu/8.0.0/../../../../riscv32-unknown-linux-gnu/bin/ld: final link failed: Bad value
collect2: error: ld returned 1 exit status
../Makerules:745: recipe for target '/scratch/palmer/work/upstream/riscv-gnu-toolchain/build/build-glibc-linux-32/libc.so' failed
make[2]: *** [/scratch/palmer/work/upstream/riscv-gnu-toolchain/build/build-glibc-linux-32/libc.so] Error 1
make[2]: Leaving directory '/scratch/palmer/work/upstream/riscv-gnu-toolchain/riscv-glibc/elf'
Makefile:215: recipe for target 'elf/subdir_lib' failed
make[1]: *** [elf/subdir_lib] Error 2
make[1]: Leaving directory '/scratch/palmer/work/upstream/riscv-gnu-toolchain/riscv-glibc'
Makefile:9: recipe for target 'all' failed
````

palmer-dabbelt, 2017-05-22:
I bisected this down to the following commit: https://github.com/riscv/riscv-binutils-gdb/commit/d7153c4ac333c9127c18efcf184607d368bb142d

````
commit d7153c4ac333c9127c18efcf184607d368bb142d
Author: Alan Modra <amodra@gmail.com>
Date:   Thu Apr 27 11:08:14 2017 +0930

    Edit .eh_frame symbols
    
    Experimental support for moving symbols defined in .eh_frame as their
    CIEs/FDEs are edited or merged.
    
        * elf-bfd.h (struct eh_cie_fde): Add aug_str_len and aug_data_len.
        (_bfd_elf_adjust_eh_frame_global_symbol): Declare.
        * elf-eh-frame.c (_bfd_elf_parse_eh_frame): Set aug_str_len and
        aug_data_len.
        (offset_adjust): New function.
        (_bfd_elf_adjust_eh_frame_global_symbol): Likewise.
        (adjust_eh_frame_local_symbols): Likewise.
        (_bfd_elf_discard_section_eh_frame): Call adjust_eh_frame_local_symbols
        after changing anything.  Return true if anything changed.
        * elflink.c (bfd_elf_discard_info): If .eh_frame changed, call
        _bfd_elf_adjust_eh_frame_global_symbol for globals.

:040000 040000 b023a3cdb7059b9d21c50f7beacd04236e0dedce 673f4819c2c29dee60b1cda42ac553bd5021c385 M      bfd
````

````
git bisect start
# good: [d0f744f970225f68460eb36a4975ae92a2e9495a] bfd: RISC-V: relax to gp in more cases.
git bisect good d0f744f970225f68460eb36a4975ae92a2e9495a
# bad: [b7e0fc3067b8c33ebc3106139ef2fa173c57226f] (WIP) RISC-V: Relax RISCV_PCREL_* to RISCV_GPREL_*
git bisect bad b7e0fc3067b8c33ebc3106139ef2fa173c57226f
# good: [8473b4472dce9ce87aaad35ff8e975e1487f914e] windows-nat.c: Fix bad initialization of ptid
git bisect good 8473b4472dce9ce87aaad35ff8e975e1487f914e
# bad: [1f78f649e868979d36490d7c2ea0b3d41abc5ec6] i386: Simplify VxWorks for non-PIC
git bisect bad 1f78f649e868979d36490d7c2ea0b3d41abc5ec6
# good: [4895cde29761f6c39fdb6c319ff4666ee39f5d28] gdb_xml_parser: make data fields private and make more functions methods
git bisect good 4895cde29761f6c39fdb6c319ff4666ee39f5d28
# good: [1587442d37ee4266e54d59bfdc783574f0587aff] i386: Set ELF_MAXPAGESIZE to 0x1000 for VxWorks
git bisect good 1587442d37ee4266e54d59bfdc783574f0587aff
# good: [19c4559475791cd70e319eaf5c1e23d23d9ae81e] nds32: Abort instead of returning REG_UNKNOWN
git bisect good 19c4559475791cd70e319eaf5c1e23d23d9ae81e
# good: [ce8ad8721313d288a05a95b62d95ca43db584ebb] MIPS/GAS: Fix `.option picX' handling with relaxation
git bisect good ce8ad8721313d288a05a95b62d95ca43db584ebb
# bad: [76c20d54ca5042e11af9ddf5723cc24cf47736ec] Constify elf_backend_eh_frame_address_size
git bisect bad 76c20d54ca5042e11af9ddf5723cc24cf47736ec
# good: [21d68fcd0c2f9113369d860ee1e5841bfacc35ff] PowerPC undefweak handling
git bisect good 21d68fcd0c2f9113369d860ee1e5841bfacc35ff
# bad: [d7153c4ac333c9127c18efcf184607d368bb142d] Edit .eh_frame symbols
git bisect bad d7153c4ac333c9127c18efcf184607d368bb142d
# good: [641338d8e9b62bde9672f13f0dca2324b61e46e2] Clear dynstr_index when forcing symbols local
git bisect good 641338d8e9b62bde9672f13f0dca2324b61e46e2
# first bad commit: [d7153c4ac333c9127c18efcf184607d368bb142d] Edit .eh_frame symbols
````

aswaterman, 2017-05-22:
@kito-cheng any changes made to NDS32 port as a result of this change?

kito-cheng, 2017-05-23:
I've call @kuanlinchentw take a look at this issue, and our software version policy is based on stable release if possible, so we don't hit this yet.

kuanlinchentw, 2017-06-01:
In Alan's commit d7153c4ac333c9127c18efcf184607d368bb142d, the st_value of symbols in .eh_frame are recalculated.  It makes FDE's R_RISCV_SUB32 symbol's value be subtracted twice in adjust_eh_frame_local_symbols and _bfd_elf_write_section_eh_frame.  NDS32 port in DIFF_EXPR is different from RISCV, so the commit is ok to NDS32.  I will try to fix this issue as soon as possible.

aswaterman, 2017-06-01:
thanks for following up @kuanlinchentw; I envy your BFD expertise.

lu-ping, 2017-06-20:
just FYI, maybe you have known about it,  I observed same error on binutils upstream master branch today. the 2.28 is good.

jim-wilson, 2017-11-30:
I reproduced this, and confirmed that the patch in issue #83 fixes it.  So closing.

closed 2017-11-30 by jim-wilson