riscvarchive/riscv-binutils-gdb #77: RiscV assembler - problems with  'SLLI/SRLI/SRAI' commands for RV32I

riscCoder, 2017-05-11:
I am seeing a problem with the SLLI/SRLI/SRAI commands for a RV32I target. .  I have been 
communicating with Palmer Dabbelt - he suggested I post this as an issue/bug.

What follows are primitive test cases.

RV32I assembly source file 
```
> cat test.s

.text
        slli    x31,x31,63
        srli    x31,x31,63
        srai    x31,x31,63
```

assemble in the installed RiscV tools /bin directory 
> ./riscv32-unknown-elf-as test.s -o test.o
test.s: Assembler messages:
test.s:2: Warning: Improper shift amount (63)
test.s:3: Warning: Improper shift amount (63)
test.s:4: Warning: Improper shift amount (63)

create a list file

> ./riscv32-unknown-elf-objdump -h -l -M numeric,no-aliases -S -d -EL test.o 

```
test.o:     file format elf32-littleriscv

Sections:
Idx Name          Size      VMA       LMA       File off  Algn
  0 .text         0000000c  00000000  00000000  00000034  2**2
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
  1 .data         00000000  00000000  00000000  00000040  2**0
                  CONTENTS, ALLOC, LOAD, DATA
  2 .bss          00000000  00000000  00000000  00000040  2**0
                  ALLOC

Disassembly of section .text:
00000000 <.text>:
   0:   03ff9f93                slli    x31,x31,0x3f
   4:   03ffdf93                srli    x31,x31,0x3f
   8:   43ffdf93                srai    x31,x31,0x3f

```

The problem is the machine code for the 3 instructions are not following the spec for RV32I.  It appears the 'shamt' field is overflowing into the higher order instruction words bits.  According to the RiscV v2.1 spec, the upper 7 bits should be 0000000, 0000000, and 0100000 respectively for SLLI,SRLI, and SRAI.

I get the exact same list file and machine code for the following source code
```
.text
        slli    x31,x31,255
        srli    x31,x31,255
        srai    x31,x31,255
```

Both source files are 'overloading' the 'shamt' field width.  My expectation is for truncation and the machine code to look like the following when programing a 63 shamt value

01ff9f93  slli  x31,x31,0x3f
01ff9f93  srli  x31,x31,0x3f
41ff9f93  srai x31,x31,0x3f

Lastly - the RiscV toolchain was built from source.  
> ./riscv32-unknown-elf-as -v
GNU assembler version 2.28.51 (riscv32-unknown-elf) using BFD version (GNU Binutils) 2.28.51.20170101

This is on a recent 2017 install of openSUSE Leap 42.2
> uname -a
Linux linux 4.4.49-16-default #1 SMP Sun Feb 19 17:40:35 UTC 2017 (70e9954) x86_64 x86_64 x86_64 GNU/Linux

cheers

Dave W.

riscCoder, 2017-05-11:
I like the idea of error and halt for wrong shift amounts {'shamt').  Machine code that does not follow the RiscV spec, should not be generated.

palmer-dabbelt, 2017-05-11:
I think this has been fixed by #78.  If it hasn't then feel free to re-open this issue.

Thanks for the bug report!

closed 2017-05-11 by palmer-dabbelt