riscvarchive/riscv-binutils-gdb #82: error: ‘scm_new_port_table_entry’ was not declared in this scope

promach, 2017-06-01:
I am having build errors regarding riscv-binutils-gdb

The errors are similar to the one below;

```
/home/promach/Documents/Grive/Personal/CPU/RISC-V/rocket-chip/riscv-tools/riscv-gnu-toolchain/build/../riscv-binutils-gdb/gdb/guile/scm-ports.c:139:10: error: ‘scm_new_port_table_entry’ was not declared in this scope
   port = scm_new_port_table_entry (port_type);
          ^~~~~~~~~~~~~~~~~~~~~~~~
```

I have attached the error log file in error.txt below.

[error.txt](https://github.com/riscv/riscv-binutils-gdb/files/1045374/error.txt)

My system configuration is as follows:
Arch Linux, gcc (GCC) 7.1.1 20170528 , GNU ld (GNU Binutils) 2.28.0.20170506

aswaterman, 2017-06-01:
This isn't a problem in binutils; rather, it is a Guile library version mismatch.  See https://github.com/riscv/riscv-gnu-toolchain/issues/244 and https://github.com/riscv/riscv-gnu-toolchain/pull/247

closed 2017-06-01 by aswaterman

kyrias, 2017-06-02:
Specifically the workaround with riscv/riscv-gnu-toolchain#247 would be to either install a compatible Guile version and specify that, or specify `--with-guile=no`, when building riscv-gnu-toolchain.

promach, 2017-06-03:
@kyrias Strange, I do not understand why is your pull request not merged.  

I have tried with ./build.sh --with-guile=no , I suppose this is not the correct way to do so.

Do I need to nodify the 3 files in riscv-gnu-toolchain on my own ?

promach, 2017-06-03:
@kyrias sorry for multiple posting of comments. I am now usng the merged commit and the error still persists.  I have tried building toolchain with and without using autoreconf, I have no idea what I have missed out. Do you have any idea ?

kyrias, 2017-06-03:
Add `--with-guile=no` to the riscv-gnu-toolchain line in build.sh.

promach, 2017-06-03:
`#! /bin/bash
#
# Script to build RISC-V ISA simulator, proxy kernel, and GNU toolchain.
# Tools will be installed to $RISCV.

. build.common

echo "Starting RISC-V Toolchain build process"

build_project riscv-openocd --prefix=$RISCV --enable-remote-bitbang --enable-jtag_vpi --disable-werror
build_project riscv-fesvr --prefix=$RISCV
build_project riscv-isa-sim --prefix=$RISCV --with-fesvr=$RISCV
build_project riscv-gnu-toolchain --prefix=$RISCV --with-guile=no
CC= CXX= build_project riscv-pk --prefix=$RISCV --host=riscv64-unknown-elf
build_project riscv-tests --prefix=$RISCV/riscv64-unknown-elf

echo -e "\\nRISC-V Toolchain installation completed!"`

also does not work. 

My system guile version is      guile (GNU Guile) 2.2.2

promach, 2017-06-03:
adding that to build.sh also does not work. 
My system guile version is      guile (GNU Guile) 2.2.2

#! /bin/bash
#
# Script to build RISC-V ISA simulator, proxy kernel, and GNU toolchain.
# Tools will be installed to $RISCV.

. build.common

echo "Starting RISC-V Toolchain build process"

build_project riscv-openocd --prefix=$RISCV --enable-remote-bitbang --enable-jtag_vpi --disable-werror
build_project riscv-fesvr --prefix=$RISCV
build_project riscv-isa-sim --prefix=$RISCV --with-fesvr=$RISCV
build_project riscv-gnu-toolchain --prefix=$RISCV --with-guile=no
CC= CXX= build_project riscv-pk --prefix=$RISCV --host=riscv64-unknown-elf
build_project riscv-tests --prefix=$RISCV/riscv64-unknown-elf

echo -e "\\nRISC-V Toolchain installation completed!"

kyrias, 2017-06-03:
Are you entirely certain that the riscv-gnu-toolchain submodule is in fact the one with it?  And what exactly does the build.sh line look like now?

kyrias, 2017-06-05:
@promach Could you try running `git -C riscv-gnu-toolchain describe --long --tags` from the riscv-tools directory?

promach, 2017-06-05:
@kyrias Thanks. It is solved.