riscvarchive/riscv-binutils-gdb #85: Unrecognized opcode error received for fsrmi and fsflagsi

mneilly, 2017-06-13:
The fsrmi and fsflagsi pseudo-ops defined in the 2.2 RISC-V ISA manual are not recognized with the latest toolchain. The register forms of the pseudo-ops compile.

Attempting to compile the following code:

```

#include <stdio.h>

int main(int argc, char **argv)
{
    int rd, rs;

    asm volatile("frrm %0" : "=r"(rd));
    asm volatile("fsrm %0, %1" : "=r"(rd) : "r"(rs));
    asm volatile("fsrm %0" : : "r"(rs));
    asm volatile("fsrmi %0, %1" : "=r"(rd) : "i"(0));
    asm volatile("fsrmi %0" : : "i"(0));

    asm volatile("frflags %0" : "=r"(rd));
    asm volatile("fsflags %0, %1" : "=r"(rd) : "r"(rs));
    asm volatile("fsflags %0" : : "r"(rs));
    asm volatile("fsflagsi %0, %1" : "=r"(rd) : "i"(0));
    asm volatile("fsflagsi %0" : : "i"(0));
}
```

Results in the following errors:

```
$ riscv64-unknown-elf-gcc -o /tmp/test /tmp/test-pseudo-ops.c
/tmp/test-pseudo-ops.c: Assembler messages:
/tmp/test-pseudo-ops.c:11: Error: unrecognized opcode `fsrmi a5,0'
/tmp/test-pseudo-ops.c:12: Error: unrecognized opcode `fsrmi 0'
/tmp/test-pseudo-ops.c:17: Error: unrecognized opcode `fsflagsi a5,0'
/tmp/test-pseudo-ops.c:18: Error: unrecognized opcode `fsflagsi 0'
$
```

aswaterman, 2017-06-13:
I'm actually wondering whether we want to remove these pseudo-ops from the ISA spec (they arguably don't belong there anyway) and recommend the use of e.g. ```csrwi frm, 0```, ```csrrwi rd, frm, 0``` instead...

mneilly, 2017-06-14:
No problem on my end if that's the solution. I was just looking for consistency between the docs and the tools. Since the other pseudo-ops in the ISA manual were present it seemed these should be as well.

aswaterman, 2017-06-14:
Yep, thanks for pointing out the discrepancy.  I've updated the ISA manual.

closed 2017-06-14 by aswaterman