riscvarchive/riscv-binutils-gdb #86: Failed to build glibc for riscv64

noamc, 2017-06-14:
Hi,

I am trying to work with buildroot to create the risv64 toolchain and later create my vmlinux.
My buildroot tree is based on SiFive work and can be seen at this [branch](https://github.com/Mellanox/buildroot/commits/riscv)
Here is my [patch](https://github.com/Mellanox/buildroot/commit/d22f22640bf51a03be6aa34654576694cc746e06)

Note: I use for binutils. gcc, glibc, and linux headers the riscv-next branch.

To reproduce
I do:
make mlx_riscv64_defconfig
make 

I get an error during glibc build phase similar to #84 see below log:
I tried the FIX proposed to riscv32 slightly changed, but it did not help.
The modified patch can be seen embedded in above link.

Any ideas?

/home/noam/workspace/newenv/buildroot/output/host/usr/bin/riscv64-buildroot-linux-gnu-gcc   -shared -static-libgcc -Wl,-O1  -Wl,-z,defs -Wl,-dynamic-linker=/lib/ld-linux-riscv64-lp64d.so.1  -B/home/noam/workspace/newenv/buildroot/output/build/glibc-riscv-next/build/csu/  -Wl,--version-script=/home/noam/workspace/newenv/buildroot/output/build/glibc-riscv-next/build/libc.map -Wl,-soname=libc.so.6 -Wl,-z,combreloc -Wl,-z,relro -Wl,--hash-style=both  -nostdlib -nostartfiles -e __libc_main -L/home/noam/workspace/newenv/buildroot/output/build/glibc-riscv-next/build -L/home/noam/workspace/newenv/buildroot/output/build/glibc-riscv-next/build/math -L/home/noam/workspace/newenv/buildroot/output/build/glibc-riscv-next/build/elf -L/home/noam/workspace/newenv/buildroot/output/build/glibc-riscv-next/build/dlfcn -L/home/noam/workspace/newenv/buildroot/output/build/glibc-riscv-next/build/nss -L/home/noam/workspace/newenv/buildroot/output/build/glibc-riscv-next/build/nis -L/home/noam/workspace/newenv/buildroot/output/build/glibc-riscv-next/build/rt -L/home/noam/workspace/newenv/buildroot/output/build/glibc-riscv-next/build/resolv -L/home/noam/workspace/newenv/buildroot/output/build/glibc-riscv-next/build/crypt -L/home/noam/workspace/newenv/buildroot/output/build/glibc-riscv-next/build/mathvec -L/home/noam/workspace/newenv/buildroot/output/build/glibc-riscv-next/build/support -L/home/noam/workspace/newenv/buildroot/output/build/glibc-riscv-next/build/nptl -Wl,-rpath-link=/home/noam/workspace/newenv/buildroot/output/build/glibc-riscv-next/build:/home/noam/workspace/newenv/buildroot/output/build/glibc-riscv-next/build/math:/home/noam/workspace/newenv/buildroot/output/build/glibc-riscv-next/build/elf:/home/noam/workspace/newenv/buildroot/output/build/glibc-riscv-next/build/dlfcn:/home/noam/workspace/newenv/buildroot/output/build/glibc-riscv-next/build/nss:/home/noam/workspace/newenv/buildroot/output/build/glibc-riscv-next/build/nis:/home/noam/workspace/newenv/buildroot/output/build/glibc-riscv-next/build/rt:/home/noam/workspace/newenv/buildroot/output/build/glibc-riscv-next/build/resolv:/home/noam/workspace/newenv/buildroot/output/build/glibc-riscv-next/build/crypt:/home/noam/workspace/newenv/buildroot/output/build/glibc-riscv-next/build/mathvec:/home/noam/workspace/newenv/buildroot/output/build/glibc-riscv-next/build/support:/home/noam/workspace/newenv/buildroot/output/build/glibc-riscv-next/build/nptl -o /home/noam/workspace/newenv/buildroot/output/build/glibc-riscv-next/build/linkobj/libc.so -T /home/noam/workspace/newenv/buildroot/output/build/glibc-riscv-next/build/shlib.lds /home/noam/workspace/newenv/buildroot/output/build/glibc-riscv-next/build/csu/abi-note.o /home/noam/workspace/newenv/buildroot/output/build/glibc-riscv-next/build/elf/soinit.os -Wl,--whole-archive /home/noam/workspace/newenv/buildroot/output/build/glibc-riscv-next/build/linkobj/libc_pic.a -Wl,--no-whole-archive /home/noam/workspace/newenv/buildroot/output/build/glibc-riscv-next/build/elf/sofini.os /home/noam/workspace/newenv/buildroot/output/build/glibc-riscv-next/build/elf/interp.os /home/noam/workspace/newenv/buildroot/output/build/glibc-riscv-next/build/elf/ld.so -lgcc
/home/noam/workspace/newenv/buildroot/output/host/usr/lib/gcc/riscv64-buildroot-linux-gnu/8.0.0/../../../../riscv64-buildroot-linux-gnu/bin/ld: .eh_frame_hdr refers to overlapping FDEs.
/home/noam/workspace/newenv/buildroot/output/host/usr/lib/gcc/riscv64-buildroot-linux-gnu/8.0.0/../../../../riscv64-buildroot-linux-gnu/bin/ld: final link failed: Bad value
collect2: error: ld returned 1 exit status
Makerules:763: recipe for target '/home/noam/workspace/newenv/buildroot/output/build/glibc-riscv-next/build/linkobj/libc.so' failed
make[3]: *** [/home/noam/workspace/newenv/buildroot/output/build/glibc-riscv-next/build/linkobj/libc.so] Error 1
Makefile:9: recipe for target 'all' failed
make[2]: *** [all] Error 2
package/pkg-generic.mk:227: recipe for target '/home/noam/workspace/newenv/buildroot/output/build/glibc-riscv-next/.stamp_built' failed
make[1]: *** [/home/noam/workspace/newenv/buildroot/output/build/glibc-riscv-next/.stamp_built] Error 2
Makefile:79: recipe for target '_all' failed
make: *** [_all] Error 2

palmer-dabbelt, 2017-06-14:
Unless you have a strong reason to use the "riscv-next" branch, I'd recommend using the "riscv-binutils-2.28" branch.  That is based on the stable binutils 2.28 release branch, which is a bit more suitable for distributions.  We maintain this branch with all the RISC-V patches that make sense (which we also submit upstream once they're ready).  There's a similar release branch for GCC, and there will be one for Linux and glibc once they've released upstream.

I'm going to leave this open, as it's probably a real bug.

jim-wilson, 2017-12-01:
Reproduced, and verified that this is fixed by issue #84.  So closing.

There is a misunderstanding here about address sizes.  Dwarf makes a distinction between program addresses whose size is determined by the ABI, and dwarf internal references, which are section relative offsets, fixed in size at 32-bits or 64-bits.  GCC emits 32-bit dwarf for all targets except IBM AIX which chose to standardize on 64-bit dwarf for 64-bit ABIs.  There is no need to move from 32-bit dwarf to 64-bti dwarf unless you have a single dwarf section which exceeds 4GB in size.  Since dwarf is split across multiple sections, you would need about 8GB of dwarf info total before 64-bit dwarf becomes necessary.
Also, since 64-bit dwarf is larger than 32-bit dwarf, it should not be used unless you actually require it.  So, since we have 32-bit dwarf here with a lp64 ABI, it is still only a 32-bit reloc fix that we need, which is what is in issue #84.

This does bring up an interesting point though.  If a new reloc had to be added to the ABI to make 32-bit dwarf work correctly, then we probably need a new reloc to make 64-bit dwarf work correctly too.  We may want to fix the ABI long before we actually need 64-bit dwarf.  So this should be checked, but this should be a different issue that the one reported here.

closed 2017-12-01 by jim-wilson