riscvarchive/riscv-binutils-gdb #90: riscv64 gdb fail to operate

noamc, 2017-06-17:
I am trying to work with GDB on target with no success.
see below log.
Whats the GDB project status? to what level WIP reached?

# gdb /hello 
GNU gdb (GDB) 7.12.50.20170505-git
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "riscv64-buildroot-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /hello...done.
(gdb) b main
No registers.
(gdb) c
The program is not being run.
(gdb) info b
No breakpoints or watchpoints.
(gdb) file /hello 
Load new symbol table from "/hello"? (y or n) y
Reading symbols from /hello...done.
(gdb) b main
No registers.
(gdb) run
Don't know how to run.  Try "help target".
(gdb) quit

sorear, 2017-06-17:
Don't run GDB on the target.

I've had success on several occasions with "target remote" (e.g. connecting to qemu), but never with "target native" (the default) or "target core".  I've heard "target sim" works but I haven't personally had a use for it.

rajeshkmeena, 2018-03-08:
use first following to connect remote target on gdb ...
 
(gdb) target remote localhost:3333

jim-wilson, 2018-09-28:
Linux native support now exists in upstream FSF gdb, and will appear in riscv-binutils-gdb when our local gdb port gets replaced with the FSF one.

closed 2018-09-28 by jim-wilson