riscvarchive/riscv-binutils-gdb #93: riscv-next doesn't build for me

timsifive, 2017-07-18:
```
$ CFLAGS=-Og ./configure --prefix=$RISCV --target=riscv64-unknown-elf --enable-tui --disable-werror
...
$ make
...
libtool: compile:  gcc -DHAVE_CONFIG_H -I. -I. -I. -I./../include -DHAVE_riscv_elf64_vec -DHAVE_riscv_elf32_vec -DHAVE_elf64_le_vec -DHAVE_elf64_be_vec -DHAVE_elf32_le_vec -DHAVE_elf32_be_vec -DHAVE_plugin_vec -DBINDIR=\"/opt/riscv/bin\" -W -Wall -Wstrict-prototypes -Wmissing-prototypes -Wshadow -I./../zlib -Og -MT elf64-riscv.lo -MD -MP -MF .deps/elf64-riscv.Tpo -c elf64-riscv.c -o elf64-riscv.o
./elfnn-riscv.c: In function ‘riscv_free_pcgp_relocs’:
./elfnn-riscv.c:2752:3: error: ‘for’ loop initial declarations are only allowed in C99 or C11 mode
```

What am I missing?
(I am assuming that riscv-next is the current development branch, where new development should be done.)

aswaterman, 2017-07-18:
I'm guessing that you're using GCC 4 and whoever wrote the code is using GCC 5 or newer, which changed the default to C99.  I'll PR a fix.

aswaterman, 2017-07-18:
https://github.com/riscv/riscv-binutils-gdb/pull/94

timsifive, 2017-07-19:
Now it builds. Thanks.

If you turn on Travis for this project, I'll be happy to make a .travis.yml file to make it build.

closed 2017-07-19 by timsifive