riscvarchive/riscv-binutils-gdb #97: addr2line and friends do not work correctly on relocations in object files

atraber, 2017-07-21:
Hi guys

Using the riscv-next branch of riscv-binutils-gdb (commit 9ce2a373296fc28c8cccf1c05ca7718ebbdf0f82), I noticed some strange behavior regarding addr2line on object files, although I think it is not limited to that and extends to more binutils tools that are used on object files. A word of warning, my analysis below is based on my understanding of gas, bfd and relocations, which is very limited and thus it may be completely wrong ;-)

Specifically I wanted to get the line number for an address of a previously compiled object file

    riscv32-unknown-elf-addr2line -e arch.o 0x24

which outputs

    /home/atraber/tmp/addr2line/arch.c:9 (discriminator 1)

The line number it reports is very far off of what I would expect (correct would be line 24).

I spent some time investigating this and it seems that the reason lies in the relocations that are being used for the dwarf2 line information contained in the object file. Specifically, the dwarf2 line information tries to advance the PC by a delta that consists of a symbol A - symbol B. This is first encoded as a BFD_RELOC_16, which then gets split into  BFD_RELOC_RISCV_ADD16 and BFD_RELOC_RISCV_SUB16 by gas md_apply_fixup.

Those BFD_RELOC_RISCV_ADD16 and BFD_RELOC_RISCV_SUB16 are what we find in the object file. On a final link to an elf file, those relocation seem to be performed correctly and thus addr2line works on it, but in an object file, the final fixup has not been performed yet.

addr2line uses bfd_find_nearest_line_descriminator, which uses _bfd_dwarf2_find_nearest_line and in the end calls bfd_get_relocated_section_contents. There is no target specific macro for this function, thus bfd ends up using the generic version, which is not aware of any special handling for ADD16 and SUB16 (in contrast to riscv_elf_relocate_section).
So what happens is that the HOWTO is followed directly, and thus when the two relocations ADD16 and SUB16 are applied, instead of subtracting the 16-bit value, it just replaces it in the dwarf2 debug information (src mask is 0, after all). Since the line information should have been encoded as a delta, but now is an absolute number, the line information is completely wrong.

Since I know next to nothing about bfd and relocations, I had to give up at this point. I think the problem comes from the fact that there is no target specific implementation of bfd_elfNN_bfd_get_relocated_section_contents, but this is mostly guessing on my part. I actually don't understand why there need to be 2 separate functions (get_relocated_section_contents vs. relocate_section) to do basically the same thing.

See the attached archive for a very simple example of the problem I'm seeing.

[test.tar.gz](https://github.com/riscv/riscv-binutils-gdb/files/1165266/test.tar.gz)

palmer-dabbelt, 2017-07-21:
https://github.com/riscv/riscv-gnu-toolchain/issues/248

jim-wilson, 2017-12-01:
Reproduced and confirmed that it is fixed by issue #98 which is mentioned in issue #248.  So closing.

closed 2017-12-01 by jim-wilson