riscvarchive/riscv-binutils-gdb #120: GAS should accept fmv.x.w and fmv.w.x

aswaterman, 2017-11-23:
The instructions fmv.x.w and fmv.w.x are the new names of fmv.x.s and fmv.s.x, respectively.  We do not currently accept the new names.  We should fix both the assembler and disassembler (but continue to accept the old names in the assembler).

palmer-dabbelt, 2017-11-23:
@jim-wilson do you mind doing this?  I think this one (and the lw fix) are lower priority than getting gcc-7.3.0 sane, but they're both easy time fillers so however you want to do it is fine with me.

aswaterman, 2017-11-23:
Yeah, this is low priority.  I just wanted to file the issue so we don't forget.

jim-wilson, 2017-11-23:
Yes, I can work on this and the lw fix.

jim-wilson, 2017-11-27:
The assembler already supports fmv.x.w and fmv.w.x.  This was added by Nick Clifton in late September in response to a bug report from Alex Bradbury.
https://sourceware.org/bugzilla/show_bug.cgi?id=22179

closed 2017-11-27 by jim-wilson

aswaterman, 2017-11-28:
Oops, I guess that patch didn't land on the branch I was using.

palmer-dabbelt, 2017-11-28:
I just backported it: https://github.com/riscv/riscv-binutils-gdb/commit/8481854f09fa97ae793b09ffc575fb0a9d1da019