riscvarchive/riscv-binutils-gdb #123: HINT instructions issue in RISCV GNU assembler

Nelson1225, 2017-12-13:
Hi all,

When verifying assembler (as for short) in upstream, we found that as considers HINT instruction as illegal one and outputs the error message as follows:

test.s: Assembler messages:
test.s:2: Error: illegal operands c.addi a4,0
test.s:3: Error: illegal operands c.add zero,s1

According to RISCV specification, HINT instruction should be considered as no-ops instead of illegal one. So, if this is one of TODO issues, we can do it and then send the patch out.

Thanks
Nelson Ching-Yu Chu 
Software R&D, Andes Technology

aswaterman, 2017-12-13:
@jim-wilson can you take a look at this?  @Nelson1225 is right that the assembler should accept opcodes marked as HINT.

These are the RVC HINTs from the manual:

- C.ADDI when imm=0 or when imm != 0 and rd=0
- C.NOP when imm != 0 (note, this is the same as C.ADDI when imm != 0 and rd=0)
- C.LI when rd=0
- C.LUI when rd=0 and imm != 0
- C.SRLI when imm=0
- C.SRAI when imm=0
- C.SLLI when rd=0
- C.MV when rd=0
- C.ADD when rd=0

Conversely, we should reject the following illegal opcodes:

- C.ADDI4SPN when imm=0
- C.ADDIW when rd=0
- C.ADDI16SP when imm=0
- C.LUI when imm=0
- C.JR when rs1=0

jim-wilson, 2017-12-13:
Yes, I can take a look at this next.

Nelson1225, 2017-12-13:
@aswaterman and @jim-wilson 
Thanks for your prompt reply :)

jim-wilson, 2017-12-14:
I'm looking at this, starting with the 
    C.ADDI when imm=0 or when imm != 0 and rd=0
case, and I'm finding lots of small inconsistencies which are making this a bigger task than it first appeared to be.

By the way, I don't see text that allows rd=0 for C.ADDI in the 2.2 doc version I have.  I do see that the docs in the riscv-isa-manual repository have more info, and allow C.NOP with non-zero immediate and effectively rd=0, but still doesn't mention C.ADDI with imm != 0 and rd=0 as a hint.  I don't think it is right to confuse C.ADDI and C.NOP this way without an explicit mention in the docs that this is allowed.  So I'm only allowing a zero immediate for C.ADDI, and not allowing the imm != 0 and rd=0 case.

As for the assembler inconsistencies, given
	.option rvc
	c.andi a4,0
	andi a4,a4,0
the assembler accepts the first one but not the second one, because the first one uses "Co" to match the immediate which does accept 0, and the second one uses "Cj" which does not accept 0.  That seems to be a simple mistake.

There are two patterns to match "li" as a compressed instruction, one which takes "Cj" which doesn't allow 0, and one which uses "0" which is intended for AMO load/store displacements, which are required to be zero, but it is odd to use it here as "0" also does load/store specific checks which make no sense for c.li.  This could just be a single line using "Co".

"Cj" was only correct for c.addi, before 0 immediates were allowed as a hint, but now that they are, Cj and Co are identical, so it makes sense to drop Co and make all of the patterns use Cj.

But the oddest thing I've found so far is that the assembler is accepting a 4 operand add instruction, where the fourth operand is required to be a constant immediate 0.  This is also using the AMO load/store offset "0" to match the constant immediate 0 which doesn't make much sense.   I can't see any point for this, and don't see anything in the user spec that would explain it.  I plan to remove this instruction unless someone can explain why it is there.

rohan:2375$ cat tmp2.s
	add a0, a1, a2, 0
rohan:2376$ ./as-new tmp2.s
rohan:2377$ ../binutils/objdump -dr a.out

a.out:     file format elf32-littleriscv

Disassembly of section .text:

00000000 <.text>:
   0:	00c58533          	add	a0,a1,a2
rohan:2378$ 

My simple patch to fix the first case is now a 7KB patch including testcases, and I will have to do some testing to verify it.

aswaterman, 2017-12-18:
Sounds like a bit of a mess. It’s probably my fault, sorry.

I suggest making sure there is no compression regression as part of your verification. Compiling something like the 403.gcc spec benchmark is a good sanity check.

jim-wilson, 2017-12-20:
>C.ADDI when imm=0 or when imm != 0 and rd=0
>C.NOP when imm != 0 (note, this is the same as C.ADDI when imm != 0 and rd=0)

The ISA manual doesn't say that c.addi is OK with rd=0 and imm != 0, so I didn't implement that.  I realize that c.nop and c.addi are effectively the same opcode, but if it isn't specified in the manual I don't think that I should implement it that way.

>C.SRLI when imm=0
>C.SRAI when imm=0

The ISA manual says that c.srli and c.srai require a non-zero immediate, but that c.srli64 and c.srai64 are hints in rv32/rv64 mode.  These are effectively the same opcode, but again it isn't specified that way in the ISA manual, so I didn't implement it that way.  Since we don't have any rv128 support yet, I didn't implement these hints.  They can be added when rv128 support is added.

The is almost ready.  I just need to do a compression regression sanity check, and then I will submit the patch.

aswaterman, 2017-12-20:
sgtm

jim-wilson, 2017-12-20:
I've added a patch with testcases to upstream binutils sources on the sourceware.org site to fix this.
https://sourceware.org/ml/binutils-cvs/2017-12/msg00131.html

closed 2017-12-20 by jim-wilson