riscvarchive/riscv-binutils-gdb #124: Linux build succeeds but Windows (cross) build fails on gdb sim

TM1234, 2017-12-24:
Overnight I tried to build the latest stuff from the repo and while the linux build completed the windows one failed:

```
x86_64-w64-mingw32-gcc -DHAVE_CONFIG_H -DWITH_DEFAULT_MODEL='"RV64G"' -DWITH_ALIGNMENT=NONSTRICT_ALIGNMENT -DWITH_TARGET_WORD_BITSIZE=64 -DWITH_TARGET_BYTE_ORDER=BFD_ENDIAN_LITTLE   -DDEFAULT_INLINE=0   -w      -I. -I/Host/Work/riscv64-unknown-elf/riscv-binutils-gdb.git/sim/riscv -I../common -I/Host/Work/riscv64-unknown-elf/riscv-binutils-gdb.git/sim/riscv/../common -I../../include -I/Host/Work/riscv64-unknown-elf/riscv-binutils-gdb.git/sim/riscv/../../include -I../../bfd -I/Host/Work/riscv64-unknown-elf/riscv-binutils-gdb.git/sim/riscv/../../bfd -I../../opcodes -I/Host/Work/riscv64-unknown-elf/riscv-binutils-gdb.git/sim/riscv/../../opcodes  -Wno-unknown-warning-option -Wno-extended-offsetof -Wno-deprecated-declarations -Wno-incompatible-pointer-types-discards-qualifiers -Wno-implicit-function-declaration -Wno-parentheses -Wno-format-nonliteral -Wno-shift-count-overflow -Wno-constant-logical-operand -Wno-shift-negative-value -Wno-format -pipe -ffunction-sections -fdata-sections -D__USE_MINGW_ACCESS -static-libstdc++ -static-libgcc -L/tmp/install/win64/lib -Wl,--gc-sections -Wl,--stack,12582912 -o run.exe \
  nrun.o libsim.a ../../bfd/libbfd.a ../../opcodes/libopcodes.a  ../../libiberty/libiberty.a -L../../zlib -lz  
libsim.a(sim-main.o):sim-main.c:(.text$execute_i+0x365b9): undefined reference to `link'
collect2: error: ld returned 1 exit status
make[3]: *** [run.exe] Error 1
Makefile:272: recipe for target 'run.exe' failed
make[3]: Leaving directory '/tmp/build/win64/binutils-gdb/sim/riscv'
make[2]: *** [all] Error 1
Makefile:129: recipe for target 'all' failed
make[2]: Leaving directory '/tmp/build/win64/binutils-gdb/sim'
make[1]: *** [all-sim] Error 2
Makefile:9539: recipe for target 'all-sim' failed
make[1]: Leaving directory '/tmp/build/win64/binutils-gdb'
make: *** [all] Error 2
Makefile:856: recipe for target 'all' failed
```

It seems odd that linux completed but windows failed but, on Friday, I had the same experience with this issue where the linux build completed and the windows one did not:

https://github.com/riscv/riscv-openocd/issues/150

Anyway for the RISC-V gcc tools I am using a slightly modified version of Liviu's GNU MCU Eclipse build process and scripts which includes cross compilation from a linux docker image for Windows:

https://github.com/gnu-mcu-eclipse/riscv-none-gcc-build

And I am telling it to build the following:

```
export NEWLIB_GIT_URL=https://github.com/riscv/riscv-newlib.git
export NEWLIB_GIT_BRANCH=riscv-newlib-2.5.0
export NEWLIB_GIT_COMMIT=32a3de0bba1535fc1ca0d8dfae147d1dacaf0979
```

Any ideas?
Thanks.

ilg-ul, 2017-12-24:
the problem seem different from https://github.com/riscv/riscv-openocd/issues/150

at first sight it looks like a missing library.

it would be easier if we could compare to an _official_ build, preferable that runs on Travis to validate each commit. 

is there such a Travis script, in some other repo? (since here I could not find one).

TM1234, 2017-12-24:
> the problem seem different from riscv/riscv-openocd#150

Yes - I didn't mean to imply that the two issues were the same but was just pointing out another case where, strangely in that case, the linux build worked but the windows one did not.

jim-wilson, 2017-12-24:
link is a POSIX syscall.  mingw of course does not support it.

Kito  added some syscall emulation code to the RISC-V simulator so that it could successfully run more programs.  One of these new syscalls is for link.  There are a few things we can do here.  We can ifdef out the link support when built on mingw.  We can add a routine to libiberty that emulates the POSIX link function for windows.  We can remove the syscall support from the simulator, and declare that the BSP (i.e. libgloss) must provide this functionality.  Or we could just drop support for link and let programs that call it fail.

The easy one is to ifdef out the code that calls link on Windows.

```
diff --git a/sim/riscv/sim-main.c b/sim/riscv/sim-main.c
index 5200c86445..444e4b829f 100644
--- a/sim/riscv/sim-main.c
+++ b/sim/riscv/sim-main.c
@@ -1764,6 +1764,7 @@ execute_i (SIM_CPU *cpu, unsigned_word iw, const struct riscv_opcode *op)
 
          switch (cpu->a7)
            {
+#ifndef __MINGW32__
            case TARGET_SYS_link:
              {
                char oldpath[1024], newpath[1024];
@@ -1772,6 +1773,7 @@ execute_i (SIM_CPU *cpu, unsigned_word iw, const struct riscv_opcode *op)
                cpu->a0 = link (oldpath, newpath);
                break;
              }
+#endif
            case TARGET_SYS_brk:
              {
                /* FIXME: Check the invalid access.  */
```

TM1234, 2017-12-24:
Thanks a lot Jim - I'll make that mod locally and try a new build.
Presumably one of the options that you outline should be chosen and applied to ensure that things build on windows/mingw (cross build from linux in my case) again?

jim-wilson, 2018-01-03:
Patch committed.

closed 2018-01-03 by jim-wilson