riscvarchive/riscv-binutils-gdb #134: GDB building with TUI support

sherrellbc, 2018-01-11:
I was developing on an Arch Linux machine for some time and was using the GDB binary compiled together when building [riscv-tools](https://github.com/riscv/riscv-tools/blob/master/README.md). However, I have recently needed to begin working from an Ubuntu machine. So, I set forth and recloned/compiled riscv-tools after installing the necessary packages (see the riscv-tools link). However, only this time GDB was compiled without Text User Interface ([TUI](https://sourceware.org/gdb/onlinedocs/gdb/TUI-Commands.html)) support (e.g. _layout_, _focus_).

It seems likely that the lack of certain packages during configuration resulted in some subset of features being disabled. What are the required packages/dependencies so that GDB will build with TUI support?

ilg-ul, 2018-01-11:
it might not be exactly your case, but in some of my GNU MCU Eclipse builds I noticed GDB to have a reference to ncurses.so.

sherrellbc, 2018-01-11:
@ilg-ul Thanks for responding so quickly. I figured it was related to curses, but I made sure to have that package installed (`libncurses5`) on Ubuntu. 

I've been poking around a bit in `configure.ac` and found a few things:

Should TUI be included at all:
https://github.com/riscv/riscv-binutils-gdb/blob/5852e73c72dbb9b86daf36c5c8afb26f2d292ef3/gdb/configure.ac#L558-L561

If so, check to see if curses is installed:
https://github.com/riscv/riscv-binutils-gdb/blob/5852e73c72dbb9b86daf36c5c8afb26f2d292ef3/gdb/configure.ac#L563-L577

Include TUI source:
https://github.com/riscv/riscv-binutils-gdb/blob/5852e73c72dbb9b86daf36c5c8afb26f2d292ef3/gdb/configure.ac#L579-L596

This seems to be the only dependency, but it doesn't seem to find my installation. Perhaps it is located in a directory that is not searched, like one of the comments alludes to.

ilg-ul, 2018-01-11:
check `config.log`, perhaps you can spot something.

alternately you can start configure with `bash -x configure ...` and you will see exactly what the script is doing.

sherrellbc, 2018-01-11:
I did not find anything in `config.log` (notably `ac_cv_search_waddstr`). This tag does not appear at all, actually. 

I tried `./configure --enable-tui` and was met with `configure: error: no enhanced curses library found; disable TUI` when building. 

I must misunderstand the process as I was expecting this to be an error at "configure" time, not compile-time.

jim-wilson, 2018-01-11:
You need the libncurses5-dev package.  On ubuntu, libncurses5 contains the libraries necessary to run programs compiled with curses.  libncurses5-dev contains the packages necessary to compile programs with curses.  Installing the -dev package installs both of them.

ilg-ul, 2018-01-11:
Ah, Jim is right, for each library required by the build you need the headers and possibly other files, which are usually in a corresponding `-dev` package.

sherrellbc, 2018-01-12:
That was it. Thanks for the help.

closed 2018-01-12 by sherrellbc

sherrellbc, 2018-01-12:
Unfortunately other features seem to not work as well now. For instance, `next`, or `ni` seems to behave exactly like `step`/`si`.