riscvarchive/riscv-binutils-gdb #138: ld mislinks PCREL_LO12_I with addend

Bigcheese, 2018-02-10:
I've produced [this object file](https://github.com/riscv/riscv-binutils-gdb/files/1713027/test-llvm.o.gz) using a patched llvm from the following assembly:
```
  .text
1:
  auipc ra, %pcrel_hi(tdata)
  addi ra, ra, %pcrel_lo(1b)
  lb t1, 0(ra)
foo:
1:
  auipc ra, %pcrel_hi(tdata)
  addi ra, ra, %pcrel_lo(1b)
  lb t2, 1(ra)

  .data
tdata:
  .byte 0xff
  .byte 0x00
  .byte 0xf0
  .byte 0x0f
```

which generates an object file which looks like:
```
Disassembly of section .text:

00000000 <foo-0xc>:
   0:   00000097                auipc   ra,0x0
                        0: R_RISCV_PCREL_HI20   .data
   4:   00008093                mv      ra,ra
                        4: R_RISCV_PCREL_LO12_I .text
   8:   00008303                lb      t1,0(ra) # 0 <foo-0xc>

0000000c <foo>:
   c:   00000097                auipc   ra,0x0
                        c: R_RISCV_PCREL_HI20   .data
  10:   00008093                mv      ra,ra
                        10: R_RISCV_PCREL_LO12_I        .text+0xc
  14:   00108383                lb      t2,1(ra) # d <foo+0x1>
```

When I link it with riscv-ld I get:

```
Disassembly of section .text:

00010000 <foo-0xc>:
   10000:       00001097                auipc   ra,0x1
   10004:       00008093                mv      ra,ra
   10008:       00008303                lb      t1,0(ra) # 11000 <tdata>

0001000c <foo>:
   1000c:       00001097                auipc   ra,0x1
   10010:       00c08093                addi    ra,ra,12 # 11018 <_end+0x14>
   10014:       00108383                lb      t2,1(ra)
```

I expect `10010: addi    ra,ra,12` to instead be `10010: addi    ra,ra,-12`

If I hex edit the R_RISCV_PCREL_LO12_I relocation at `0x10` to point to `foo` instead of `.text+0xc` it works correctly.  However llvm doesn't emit temporary labels such as `1` when it can form the same thing using a section symbol.

sorear, 2018-02-10:
Can we address it by modifying the psABI to forbid addends here?

Bigcheese, 2018-02-10:
Is this any harder to implement?  I've never touched binutils before but from looking at the [code](https://github.com/riscv/riscv-binutils-gdb/blob/riscv-binutils-2.29/bfd/elfnn-riscv.c#L3227) it seems it just needs to account for the addend when finding the PCREL_HI20 reloc.

aswaterman, 2018-02-10:
I agree the psABI should forbid this.

sorear, 2018-02-11:
To be clear I'm not opposed to this as an enhancement; it would shrink object files significantly (removing the .Lnnn symbols saves 5217 out of 19504 bytes on a random object file from libffi).  But I don't think there should have been an expectation that it would work.

palmer-dabbelt, 2018-02-13:
I'm OK forbidding this in the ABI, and we should at least be checking to make sure there's a 0 addend rather than just producing the wrong answer.

jim-wilson, 2018-02-14:
I've written a binutils patch, though I'd like to verify that I'm doing it right.  I'm only checking for an addend on R_RISCV_PCREL_LO12_{I,S}, as those are the only ones that point back into the instruction stream, and hence I think those are the only ones that are broken.  We could optionally allow this when not relaxing, I'm currently giving an error unconditionally.  I've got a testcase modified to work with gas, using %pcrel_lo(.text+12), and I get from the linker

rohan:2195$ ld/ld-new tmp.o
ld/ld-new: warning: cannot find entry symbol _start; defaulting to 0000000000010074
tmp.o: In function `foo':
(.text+0x10): dangerous relocation: %pcrel_lo with addend
rohan:2196$ echo $?
1
rohan:2197$

sorear, 2018-02-14:
I think it should forbid all relocations with a nonzero addend where the target symbol is in a relaxable section, if that's something you can check with a simple pointer chase?

jim-wilson, 2018-02-14:
I can try that.  This is more complicated, but as you say, should just involve some pointer chasing.

I worry it might have some side-effects.  For instance, an assembly programmer who puts some read-only data in the text section and indexes into it could get a linker error even though this should work fine.

sorear, 2018-02-14:
So we have two overlapping issues:

1. LO12 relocations are completely ignoring the addend without any diagnostic

2. relocations with addends pointing to text may point to the wrong thing after linking

I was mixing these up earlier; what you were doing was correct for 1.

jim-wilson, 2018-02-15:
I committed a patch to upstream binutils that gives an error for this testcase.

jim-wilson, 2018-09-28:
Was fixed in binutils, and that patch has been merged into riscv-binutils-gdb.

closed 2018-09-28 by jim-wilson