riscvarchive/riscv-binutils-gdb #141: GDB using compressed breakpoint instructions for s/w BPs after setting no compression?

rajeshkmeena, 2018-03-07:
" set riscv use_compressed_breakpoints no " after using this GDB is compressing breakpoint instruction.

I try this command after setting path for compiler and as well as before launching gdb , but its not working . Any reason ??

timsifive, 2018-03-07:
From code inspection, it looks like valid options for this setting are: "on", "off", "auto"

Can you try: `set riscv use_compressed_breakpoints off`

rajeshkmeena, 2018-03-08:
When I launch GDB with default mode ...its working and setting compressed breakpoint....

(gdb).....
.....XML support was disabled at compile time
0x0000000020000004 in main () at test.c:9
9	    while (a==0)
(gdb) disassemble
Dump of assembler code for function main:
   0x0000000020000000 <+0>:	lui	a5,0x20000
=> 0x0000000020000004 <+4>:	ld	a5,88(a5) # 0x20000058 <a>
   0x0000000020000008 <+8>:	beqz	a5,0x20000000 <main>
   0x000000002000000a <+10>:	lui	a5,0x20000
   0x000000002000000e <+14>:	li	a4,1
   0x0000000020000010 <+16>:	sd	a4,80(a5) # 0x20000050 <c>
   0x0000000020000014 <+20>:	lui	a5,0x20000
   0x0000000020000018 <+24>:	ld	a4,88(a5) # 0x20000058 <a>
   0x000000002000001c <+28>:	li	a5,1
   0x000000002000001e <+30>:	beq	a4,a5,0x20000014 <main+20>
   0x0000000020000022 <+34>:	lui	a5,0x20000
   0x0000000020000026 <+38>:	ld	a4,80(a5) # 0x20000050 <c>
   0x000000002000002a <+42>:	li	a5,1
   0x000000002000002c <+44>:	beq	a4,a5,0x20000022 <main+34>
   0x0000000020000030 <+48>:	lui	a5,0x20000
   0x0000000020000034 <+52>:	sd	zero,88(a5) # 0x20000058 <a>
   0x0000000020000038 <+56>:	lui	a5,0x20000
   0x000000002000003c <+60>:	ld	a5,88(a5) # 0x20000058 <a>
   0x0000000020000040 <+64>:	beqz	a5,0x20000038 <main+56>
   0x0000000020000042 <+66>:	lui	a5,0x20000
   0x0000000020000046 <+70>:	li	a4,1
   0x0000000020000048 <+72>:	sd	a4,80(a5) # 0x20000050 <c>
---Type <return> to continue, or q <return> to quit---
   0x000000002000004c <+76>:	li	a0,0
   0x000000002000004e <+78>:	ret
End of assembler dump.
**(gdb) b 11
Breakpoint 1 at 0x2000000a: file test.c, line 11.**
(gdb) c
Continuing.
^C
Program received signal SIGINT, Interrupt.
0x0000000020000004 in main () at test.c:9
9	    while (a==0)
(gdb) print a =1
$1 = 1
(gdb) c
Continuing.

Breakpoint 1, main () at test.c:13
13	     b = 1;
(gdb) ........

From Openocd 
Debug: 9389 61603 program.c:34 riscv_program_write(): 0x7fffcb01d870: debug_buffer[00] = lh      s1, 0(s0)
Debug: 9392 61603 program.c:34 riscv_program_write(): 0x7fffcb01d870: debug_buffer[01] = addi    s0, s0, 2
Debug: 9791 64311 program.c:34 riscv_program_write(): 0x7fffcb01d830: debug_buffer[00] = lh      s1, 0(s0)
Debug: 9792 64311 riscv-013.c:265 scan(): 40b w 00041483 @20 -> + 00000000 @16
Debug: 9793 64311 riscv-013.c:265 scan(): 40b - 00000000 @20 -> + 00041483 @20
Debug: 9794 64311 program.c:34 riscv_program_write(): 0x7fffcb01d830: debug_buffer[01] = addi    s0, s0, 2
Debug: 9917 64318 program.c:34 riscv_program_write(): 0x7fffcb01d7f0: debug_buffer[00] = sh      s1, 0(s0)
Debug: 9918 64318 riscv-013.c:265 scan(): 40b w 00941023 @20 -> + 00000000 @04
Debug: 9919 64318 riscv-013.c:265 scan(): 40b - 00000000 @20 -> + 00941023 @20
Debug: 9920 64318 program.c:34 riscv_program_write(): 0x7fffcb01d7f0: debug_buffer[01] = addi    s0, s0, 2
Debug: breakpoints.c:104 breakpoint_add_internal(): added software breakpoint at 0x2000000a of length 0x00000002, (BPID: 0)

But , when I launch GDB with  " **set riscv use_compressed_breakpoints off** " its failed to set breakpoint and using compressed instruction to read data from breakpoint....

(gdb) .....
.......XML support was disabled at compile time
0x0000000020000004 in main () at test.c:9
9	    while (a==0)
**(gdb) b 11
Breakpoint 1 at 0x2000000a: file test.c, line 11.
(gdb) c         
Continuing.
Warning:
Cannot insert breakpoint 1.
Cannot access memory at address 0x2000000a**

Command aborted.
(gdb) print a =1
$1 = 1
(gdb) c
Continuing.
Warning:
Cannot insert breakpoint 1.
Cannot access memory at address 0x2000000a

Command aborted.
(gdb).......... 

From Openocd 

**Debug: 6520 38218 program.c:34 riscv_program_write(): 0x7ffef3d457d0: debug_buffer[00] = lh      s1, 0(s0)
Debug: 6523 38218 program.c:34 riscv_program_write(): 0x7ffef3d457d0: debug_buffer[01] = addi    s0, s0, 2**
...
Debug: 10239 74026 program.c:34 riscv_program_write(): 0x7ffef3d45790: debug_buffer[00] = lw      s1, 0(s0)
Debug: 10242 74026 program.c:34 riscv_program_write(): 0x7ffef3d45790: debug_buffer[01] = addi    s0, s0, 4

Debug: 10268 74028 riscv-013.c:576 execute_abstract_command(): command 0x361009 failed; abstractcs=0x2000302
Debug: 10269 74028 riscv-013.c:265 scan(): 40b w 00000300 @16 -> + 00000000 @16
Debug: 10270 74028 riscv-013.c:276 scan():  cmderr=3 -> 
Debug: 10271 74028 riscv-013.c:265 scan(): 40b - 00000000 @16 -> + 00000300 @16
Debug: 10272 74028 riscv-013.c:276 scan():  ->  cmderr=3
Error: 10273 74028 riscv.c:479 riscv_add_breakpoint(): Failed to read original instruction at 0x2000000a
**Error: 10274 74028 breakpoints.c:94 breakpoint_add_internal(): can't add breakpoint: unknown reason
Debug: 10275 74028 gdb_server.c:1368 gdb_error(): Reporting -4 to GDB as generic error**

( Note : I deleted unnecessary openocd data....)
1. Can you explain, after **"set riscv use_compressed_breakpoints off"** GDB is try to read data by compressed instruction.

Debug: 6520 38218 program.c:34 riscv_program_write(): 0x7ffef3d457d0: debug_buffer[00] = lh      s1, 0(s0)
Debug: 6523 38218 program.c:34 riscv_program_write(): 0x7ffef3d457d0: debug_buffer[01] = addi    s0, s0, 2

2. I put breakpoint at same location both time , they is second way  is failed.
 (Note: line 11 (b 11) is not only single line where BP is failed, there are many line where setting BP is failed)

jim-wilson, 2018-03-08:
GDB is a work in progress.  Not all features are working.

GDB is in the process of being upstreamed.  A subset of RISC-V gdb, the core target dependent support, was added to FSF GDB about 2 days ago.  This is a modified version of what is in this repo.  I expect all future work will be in the FSF GDB tree, and this version of the sources may be dead.  Though of course at some point we can sync with the FSF GDB tree and bring that into this tree.  It may be a while before that happens.  The person doing the FSF GDB work may not be following this repo, so problems reported here may not get fixed.

There are two separate issues here.
1) Whether compressed instructions are supported or not.
2) Whether compressed breakpoint instructions are supported.
The "set riscv use_compressed_breakpoints X" only affects the second one.  The first one is affected by the architecture setting.  If you want to turn compressed instruction support off, you need to specify an architecture without compressed instruction support.  Gdb tries to get the architecture info by reading misa, or you can specify it with a command.

The address is question is 2-byte aligned.  There might be some bug preventing us from writing a 4-byte breakpoint instruction to a 2-byte aligned address.  This might be a gdb bug, or it might be a OpenOCD bug, or I suppose it could even be a simulator/hardware bug.

rajeshkmeena, 2018-04-05:
@jim-wilson thanks

closed 2018-04-05 by rajeshkmeena