riscvarchive/riscv-binutils-gdb #142: csrw requires absolute expression

vanjoe, 2018-03-12:
The above error is issued by the assembler when a csr name is not supported by the toochain in use. Is it possible to make this clearer which specific name is causing the issue. The line number is not particularly useful when the line contains a macro that has a bunch of csr instructions in it.

jim-wilson, 2018-03-15:
The assembler in riscv-binutils-gdb is out-of-date.  Eventually we will update it, but we haven't found time to do the work yet.  The assembler in the FSF Binutils 2.30 release should already support all CSRs in 1.9.1 and 1.10 of the priv spec and is a better one to use.  Though of course more CSRs will be added in the future, and the message is a bit confusing, so I think it is still worth fixing.

I'm working on a patch that does this:

gamma05:2190$ cat tmp.s
	csrr a0,random
gamma05:2191$ ./as-new tmp.s
tmp.s: Assembler messages:
tmp.s:1: Error: unknown CSR `random'
gamma05:2192$

vanjoe, 2018-03-15:
Upgrading to 2.30 did solve the problem I was having. If this repo is out of date is this the wrong spot to ask questions like this?

I'm closing because apparently it's being worked on thanks.

closed 2018-03-15 by vanjoe