riscvarchive/riscv-binutils-gdb #147: User-defined Commands are not supported

ivankravets, 2018-05-01:
See https://sourceware.org/gdb/onlinedocs/gdb/Define.html

# How to reproduce

1. Create somewhere the file named as `mydef.gdb` with the next content:
```
define HelloWorld
 	echo HelloWorld\n
end

HelloWorld
```

2. Then
```sh
> bin/riscv64-unknown-elf-gdb
(gdb) source mydef.gdb
mydef.gdb:5: Error in sourced command file:
Undefined command: "HelloWorld".  Try "help".

(gdb) HelloWorld
Undefined command: "HelloWorld".  Try "help".
```

jim-wilson, 2018-05-01:
This is a case problem.

(gdb) source mydef.gdb 
(gdb) help user
User-defined commands.
The commands in this class are those defined by the user.
Use the "define" command to define a command.

List of commands:

helloworld -- User-defined

Type "help" followed by command name for full documentation.
Type "apropos word" to search for commands related to "word".
Command name abbreviations are allowed if unambiguous.
(gdb) 

Note that the user command is lowercase.  This was a deliberate choice when this feature was added to gdb.  They later decided to allow uppercase in user defined commands.  See 
    https://sourceware.org/bugzilla/show_bug.cgi?id=21827
Our gdb sources are based on a snapshot from July 2017, and do not have this change which was made in August 2017.

closed 2018-05-01 by jim-wilson

ivankravets, 2018-05-01:
Thanks for the info!