riscvarchive/riscv-binutils-gdb #153: info->insn_type not set on disassembled instructions

bluewww, 2018-07-12:
`riscv_disassemble_insn` in `riscv-dis.c` sets info->insn_info_valid = 1 indicating that info->insn_type is useful. It is expected that insn_type reveals some information about the disassembled instruction. Currently it is always set to dis_nonbranch. It looks like that this feature is just not implemented yet.

jim-wilson, 2018-07-12:
Yes, it hasn't been implemented yet, though it looks like most binutils disassemblers do not support this feature, so this looks like a low priority item to fix.

bluewww, 2018-07-12:
I think as long as this functionality is not implemented ```info->insn_info_valid``` should not be set to a nonzero value. ```disassembler_info``` explains how it should be used.

bluewww, 2018-07-31:
Thank you for implementing this feature!