riscvarchive/riscv-binutils-gdb #154: The `li` instruction cannot load the value of the symbol

wxjstz, 2018-07-26:
I need to solve a problem:
    The program's load address and execution address are not the same. 
    The program needs to move itself to the execution address.

I got two symbols through the link script:
    _program: start address of program
    _eprogram: start address of program

I write this coe. 
```
	la	a0, _program  # load address
	li	a1, _program  # execution address
	beq	a0, a1, 2f
	li	a2, _eprogram
1:
	lb	t0, 0(a0)
	sb	t0, 0(a1)
	addi	a0, a0, 1
	addi	a1, a1, 1
	blt	a1, a2, 1b
2:
```

I get a error : illegal operands `li a1,_program'.
The `li` instruction cannot load the value of the symbol.
The `la` instruction gets the relative address.
How do I get the actual address?

wxjstz, 2018-07-26:
Usually `la` can get the value of the symbol, but the value it gets is based on the PC offset, which is related to the loading position of the program.

You need to understand my needs, the address to load and execute is different, the program needs to move itself to the appropriate address.

aswaterman, 2018-07-26:
I see.  To get the load address, you can do something like

    lw t0, _load_address_of_program

...

    .data
    _load_address_of_program: .dc.a _program

aswaterman, 2018-07-26:
It's not efficient, but it's loading code, so it doesn't matter much.

closed 2018-07-26 by wxjstz