riscvarchive/riscv-binutils-gdb #156: Enabling the execution of custom0, ..., custom3 instructions from within Linux

lucashmorais, 2018-08-13:
Hi there,

I have been working for some time on the verification of a RoCC-based accelerator and would now like to do some testing from within a Linux environment. Running tests using [riscv-pk](https://github.com/riscv/riscv-pk) on Rocket Chip has been fine, but when I call the same binaries from Linux, illegal instruction traps are triggered.

My benchmarks get custom0, ..., custom3 instructions added through [rocc-software](https://github.com/IBM/rocc-software)'s macros, which Linux doesn't seem to be very happy with.

What should I do to enable the execution of these instructions from the OS? I tried adding something like

```
{"custom0","I","d,s,t",MATCH_CUSTOM0_RD_RS1_RS2,MASK_CUSTOM0_RD_RS1_RS2, match_opcode,0},
{"custom1","I","d,s,t",MATCH_CUSTOM1_RD_RS1_RS2,MASK_CUSTOM1_RD_RS1_RS2, match_opcode,0},
{"custom2","I","d,s,t",MATCH_CUSTOM2_RD_RS1_RS2,MASK_CUSTOM2_RD_RS1_RS2, match_opcode,0},
{"custom3","I","d,s,t",MATCH_CUSTOM3_RD_RS1_RS2,MASK_CUSTOM3_RD_RS1_RS2, match_opcode,0},
```

to `riscv_opcodes` in /opcodes/riscv-opc.c but that made me get `Error: internal: bad RISC-V opcode (bits 0xfffffffffe000000 undefined): custom0 d,s,t` messages during compilation.

Is the approach involving the `.insn` that was merged with #104 useful for running custom code in Linux? Does anyone have experience with that?

Thanks a lot,
Lucas

jim-wilson, 2018-08-13:
If your RISC-V cpu can't execute the instructions you are adding, then yes, of course, you will get illegal instruction faults.  You should instead ask why it works in riscv-pk.  Maybe it has a machine mode illegal instruction handler that is catching these faults and then sending data to the accelerator.  If you want the same thing to work in linux, then you will have to modify the linux kernel trap handler to do the same thing, which may require writing a linux driver for your accelerator.  You can probably get a better answer if you ask someplace where people know what RoCC is, and know how to do linux kernel development work.

The binutils question doesn't seem relevant to your real problem, but for the build error you got, it looks like you haven't defined all instruction bits.  The assembler does a sanity check to verify that the MATCH macro ored with the MASK macro ored with all of the operand (d,s,t) field bits set to 1 should give a result of all ones for the instruction length.  In your case, some bits ended up zero, and so are missing.  These are the bits printed as ones in the error.  It looks like the upper 7 bits of the first 32-bit word are missing.  Though it is curious that it is printing a 64-bit number, are you using a 64-bit instruction format?  I don't know if that can work, as there are no currently supported 64-bit instructions.  You may have to debug the assembler a bit to see what is going on, to see if 64-bit instructions can work.

I don't know if anyone is using the .insn support.  I haven't seen any comments on it.  It was the intent that this could be using for custom instructions, though it only supports the standard 32-bit instruction formats.

seldridge, 2018-08-13:
You would need to modify Linux to set the XS bits and save/restore any RoCC state you have in the context switch code for Linux. The source of the illegal instruction is likely due to the XS bits being zero, which is analogous to having the FS bits set to zero and trying to run F/D instructions. 

Take a look at how the FPU state is saved/restored in [arch/riscv/kernel/entry.S](https://github.com/riscv/riscv-linux/blob/master/arch/riscv/kernel/entry.S). An initial (not thread safe, not secure) hack would just be to set the XS bits blindly to 1.

I don't think that this is related to binutils, though. There shouldn't be any need to recompile the kernel with binutils understanding customX. I may be missing something, though.

lucashmorais, 2018-08-15:
Thanks for the insights, I just managed to get the programs running on Linux + Spike by making the simulator do `mstatus.xs = 1` at reset. Before that, I tried recompiling linux with `sstatus` being set in the same way just after line 39 in [arch/riscv/kernel/head.S](https://github.com/riscv/riscv-linux/blob/master/arch/riscv/kernel/head.S) but for some reason the bit would have been cleared when I got to run the tests.

And of course this didn't have anything to do with binutils, in the end. 

Now that I got this easy fix I'll try to make a simple driver for doing the same. Thanks again for the help!

closed 2018-08-15 by lucashmorais