riscvarchive/riscv-binutils-gdb #157: GDB crashes on startup on RISCV-64

rohgarg, 2018-08-14:
Hi, I compiled FSF GDB (v. 8.1.1) and also top-of-tree from the
`riscv-binutils-gdb` repository directly on the QEMU RISCV64 emulator, with
Fedora-28, Linux kernel version 4.15.0-00049, and gcc-7.3.1. I used the
`./configure && make` command to configure and build.

In both of the cases, gdb fails with the following error.

```
$ gdb
arch-utils.c:694: internal-error: initialize_current_architecture: Selection of initial architecture failed
A problem internal to GDB has been detected,
further debugging may prove unreliable.
Quit this debugging session? (y or n) Segmentation fault (core dumped)
```

The kernel prints the following error trace on the terminal:

```
gdb[12303]: unhandled signal 11 code 0x1 at 0x0000000000000004 in gdb[10000+3dc000]
CPU: 1 PID: 12303 Comm: gdb Tainted: G        W        4.15.0-00049-ga3b1e7acc6a1 #28
sepc: 00000000001744cc ra : 00000000000e9744 sp : 0000003fffc61f10
 gp : 00000000003fa778 tp : 0000002000431710 t0 : 0000000000000682
 t1 : 0000000000000000 t2 : 000000000041c04f s0 : 0000000000000000
 s1 : 0000000000000004 a0 : 0000000000000000 a1 : 0000000000000002
 a2 : 0000000000000007 a3 : 0000000000455950 a4 : 5b5ad018fbf43b00
 a5 : 00000000003fb000 a6 : 0000000000000004 a7 : 0000000000000040
 s2 : 00000000003fb000 s3 : 0000000000476090 s4 : 0000000000284ee0
 s5 : 0000000000000059 s6 : 000000000000004e s7 : 00008aa090ec7820
 s8 : 0000000000000060 s9 : 00000000003fc000 s10: 000000000031b000
 s11: 00000000003fc000 t3 : 0000002000323cd2 t4 : 0000000000000000
 t5 : 0000000000000001 t6 : 0000000000000000
sstatus: 8000000000006020 sbadaddr: 0000000000000004 scause: 000000000000000d
```
(I suppose the big issue here is the unrecognized architecture, and signal 11 is just for the segfault at the end. Other programs that segfault also result in a similar error trace with signal 11.)

Is this a known issue? Any help is appreciated. 

Thanks.

jim-wilson, 2018-08-14:
riscv-binutils-gdb has a working bare metal gdb, but only if you check out the right branch.  Building riscv-gnu-toolchain will give you the right branch.  The riscv linux support is probably not working, and if it is working, it probably only works on core dumps.

GDB 8.1 doesn't have riscv support.  GDB 8.2 has only bare metal riscv support, for use with openocd.  There is a branch for it in the FSF GDB tree, but it hasn't been released yet, it should come out soon.  I just added native riscv linux support last week, so this is in top of tree gdb sources and will appear in gdb 8.3, which probably comes out early next year.  Most basic features are working, but there is a lot more work required to get a fully working gdb port.  There is no gdbserver support yet.

If it is the riscv-linux native support you are interested in, then you can build it from the gdb top of tree sources.  However, there are 3 required linux kernel patches, one optional linux kernel patch, and one optional glibc patch, none of which are in the riscv Fedora 28 distro at this time.  I don't know if or when they will be added.  The kernel is pretty easy to rebuild with patches.  Glibc is not.

You can find info on the linux kernel and glibc patches here
    https://github.com/jim-wilson/riscv-linux-native-gdb/blob/jimw-riscv-linux-gdb/README.md
and a copy of the linux kernel patches I'm currently using here
   https://github.com/jim-wilson/riscv-linux-native-gdb/blob/jimw-riscv-linux-gdb/riscv-linux.txt
But I don't recommend using this source tree anymore.  Use the FSF GDB top of tree sources instead.
    https://sourceware.org/gdb/current/

rohgarg, 2018-08-14:
Thanks for the suggestions, Jim. I'll give this a shot and see where it leads.

closed 2018-09-28 by jim-wilson