riscvarchive/riscv-binutils-gdb #161: Gdb fails to build with Python 3.7

byllgrim, 2018-10-15:
Hi.

You've got this issue, I think:
https://sourceware.org/bugzilla/show_bug.cgi?id=23252

After applying [this patch](https://sourceware.org/git/gitweb.cgi?p=binutils-gdb.git;a=blobdiff;f=gdb/python/python.c;h=20fc674f20ad6cd686976eaa879f7325ea39b17f;hp=1805c90628455be92b6788be2a531f302cef7fe9;hb=aeab512851bf6ed623d1c6c4305b6ce05e51a10c;hpb=5045b3d78903d1f6faa4cb511fa801e50dfacf48) I got past that problem.

Let me know if I did this issue reporting wrong.

jim-wilson, 2018-10-15:
We have two gdb ports, the new one is in the FSF gdb tree, and the old one is in the riscv-binutils-gdb tree.  They have slightly different feature sets and a different set of bugs.  This is a temporary problem caused by upstreaming of the gdb port which started months ago but is still incomplete.  At some point the new one will replace the old one. Meanwhile, there isn't much interest in fixing bugs in the old one.  You should try using the new one in the FSF tree instead of the old one here.

FYI There is a similar problem with riscv-glibc.  The new glibc sources can only be found in the FSF tree.    The old glibc sources are in riscv-glibc.  This is waiting for the 32-bit support to be upstreamed, and then the new sources will replace the old ones.

byllgrim, 2018-10-18:
Thank you!
I tried to look for some info like this, but the readme obviously can't tell about it, and [this page](https://github.com/riscv/riscv-binutils-gdb/wiki) was empty, and [this one](https://github.com/riscv/riscv-wiki/wiki/RISC-V-Software-Status#object-toolchain) wasn't too particular.
I'm not sure whether I should close the issue(?), as it isn't completely resolved.