riscvarchive/riscv-binutils-gdb #164: GDB : NON-STOP Mode

rajeshkrmeena, 2019-01-08:
I tried to turn on gdb non-stop mode  through three commands:
(gdb) set pagination off 
(gdb) set target-async on
(gdb) set non-stop on
But when tried to connect remote target, unable to connect
(gdb) target extended-remote localhost:3333
Remote debugging using :3333
**Non-stop mode requested, but remote does not support non-stop** (GDB message)
How can the support for non-stop mode enable for ### remote target in riscv-toolchain GDB ?

(special note : Spike-rtos.cfg used in openocd configuration for multiple riscv cores)

jim-wilson, 2019-01-08:
We have only basic gdb features implemented at present.  It will likely be a while before someone volunteers to implement the non-stop support.

rajeshkrmeena, 2019-01-09:
Thanks @jim-wilson

closed 2019-01-09 by rajeshkrmeena