riscvarchive/riscv-binutils-gdb #165: objcopy with interleave

pentin-as, 2019-01-29:
635c14e (May 18, 2018)

riscv64-unknown-elf-objcopy -I elf64-littleriscv -O verilog --interleave=16 --interleave-width=1 --byte=12 main12.hex main.elf

I have a loss of byte 0x10001a9c

main.dump:
    10001a6c:	1b050513          	addi	a0,a0,432 # 10001c18 <main+0x190>
    10001a7c:	fff00513          	li	a0,-1
    10001a8c:	00000517          	auipc	a0,0x0
    10001a9c:	0000006f          	j	10001a9c <main+0x14>

main12.hex:
@10001A6
13 13 17 
@10001AA
F0 FC 2C 58 58 F0 F0 F0 F0 F0 F0 F0 F0 F0 F0 F0 

[main.zip](https://github.com/riscv/riscv-binutils-gdb/files/2807408/main.zip)

jim-wilson, 2019-02-01:
OK.  I see the problem.  The problematic section is
  3 .text.startup 0000003c  0000000010001a64  0000000010001a64  00002a64  2**2
                  CONTENTS, ALLOC, LOAD, READONLY, CODE
the lma mod interleave is 4, which we need to account for when computing the output size.

We already have this value in "extra" and already use it for adjusting the from address and the output lma.  It looks like a simpler fix might be to just use
          size = (size + interleave - 1 - copy_byte + extra) / interleave * copy_width;
though I need to do some testing to see if that is safe.  Alternatively, we might need something like this
        if (copy_byte + extra >= interleave)
            size++;
Either one of these is simpler to understand than your patch.  This also needs an appropriate comment, and a testcase added to demonstrate the problem and verify the fix.

jim-wilson, 2019-02-04:
I now think this should be
size = (size + interleave - 1 - copy_byte + extra) / interleave * copy_width;
if (copy_byte < extra)
  size--;
but I haven't had a chance to finish this yet.