riscvarchive/riscv-binutils-gdb #170: How to display the value of floating point registers in the hex format?

hyf6661669, 2019-04-05:
I'm using ```riscv32-unknown-elf-gdb``` to debug my test codes. It is built with riscv-gnu-toolchain.

For example, I have a single-precision variable stored in the register $fa5, and its value is 2.64575124 in decimal. So the data stored in $fa5 should be 0x4029_53fd, according to IEEE 754-2008 standard. But the following commands can't display the expected result:
[image]

jim-wilson, 2019-04-05:
Try "info registers $fa5".  That should print the raw hex value of the registers along with float (and maybe double if 64-bit FP regs) values.  There are a lot of options to "info registers", you can hit the tab key to see them.

Historically, in gdb, if you use a hex format with an FP number, it first converts the FP number to an integer type and then prints that.  There is an open bug report for this.
https://sourceware.org/bugzilla/show_bug.cgi?id=16242
Though 'z' does not work anymore.  It looks like that "bug" was fixed since the problem was initially reported.

hyf6661669, 2019-04-06:
@jim-wilson 
Thanks for your fast reply, but "info registers $fa5" still prints the value of $fa5 in decimal format, which is 2.64575124. Luckily, I find an alternative method to display the value of a floating point register. In fact, the compiler trys to store a  variable ```float_temp``` in $fa5, so I can use this command to display the value in hex format:
```
(gdb) p/x float_temp
$55 = 0x2
(gdb) p/x *(int *)&float_temp
$56 = 0x402953fd
(gdb) p/r $fa5
$57 = 0x2
(gdb) 
```

However, this method seems a bit complicated. In [https://sourceware.org/gdb/onlinedocs/gdb/Output-Formats.html](url), it says GDB can only output the integer part of the value. I also spent much time on google but couldn't find related problems.

jim-wilson, 2019-04-06:
What gdb version do you have?  The current version will print values in hex.  This is on a 64-bit RISC-V linux machine.

(gdb) info registers fa0
fa0            {float = 2.64575124, 
  double = -nan(0xfffff402953fd)}	(raw 0xffffffff402953fd)
(gdb) 

The $ sign isn't necessary with info registers, I'm just so used to typing it I keep forgetting, but it is harmless and ignored.

We switched from the old riscv local gdb port to the FSF gdb port last November, so this should be working since November.  I'm not sure about before November.  I think the FSF gdb port has always printed values this way, but the old riscv local gdb port may not have been doing this.

hyf6661669, 2019-04-06:
Sorry, I find that my ```riscv-gnu-toolchain``` is out-dated. I will try to use the latest version.

hyf6661669, 2019-04-06:
@jim-wilson 
The problem is solved by using the latest riscv-gnu-toolchain, thank you!!

closed 2019-04-06 by hyf6661669