riscvarchive/riscv-binutils-gdb #176: gdb address completed

bg2d, 2019-09-29:
Hello,

I am using gdb to debug a program running on riscv.
`info files` shows me the following sections:
0x0000000000010238 - 0x0000000000010259 is .interp
0x000000000001025c - 0x000000000001027c is .note.ABI-tag
0x0000000000010280 - 0x00000000000102b0 is .hash
0x00000000000102b0 - 0x00000000000102ec is .gnu.hash
0x00000000000102f0 - 0x0000000000010398 is .dynsym
0x0000000000010398 - 0x00000000000103dd is .dynstr
0x00000000000103de - 0x00000000000103ec is .gnu.version
0x00000000000103f0 - 0x0000000000010410 is .gnu.version_r
0x0000000000010410 - 0x00000000000104a0 is .rela.plt
0x00000000000104a0 - 0x0000000000010520 is .plt
0x0000000000010520 - 0x00000000000106b6 is .text
0x00000000000106b8 - 0x00000000000106cc is .rodata
0x00000000000106cc - 0x00000000000106e0 is .eh_frame_hdr
0x00000000000106e0 - 0x000000000001070c is .eh_frame
0x0000000000011e08 - 0x0000000000011e10 is .preinit_array
0x0000000000011e10 - 0x0000000000011e18 is .init_array
0x0000000000011e18 - 0x0000000000011e20 is .fini_array
0x0000000000011e20 - 0x0000000000012000 is .dynamic
0x0000000000012000 - 0x0000000000012048 is .got
0x0000000000012048 - 0x0000000000012058 is .sdata
**0x0000000000012058 - 0x0000000000012060 is .bss**

If I try to read the .bss section from address 0x12058 I obtain:
`(gdb) x/x 0x12058`
0x12058 <completed.6023>:	0x00000000
I am expecting to read 0 but what does <completed.6023> mean?

jim-wilson, 2019-09-29:
completed.6023 is a symbol name.  The disassembler will use the closest symbol name when pretty printing data.  This may or may not be the symbol that covers the address you are looking, the disassembler doesn't know about symbol ranges, it just picks the closest one.

You probably have a variable called "completed" in your program, or the start files or libraries that you link with, and the compiled added the ".6023" to try to make it a unique variable name which won't conflict with other variable names.  This can happen a number of ways.  One such way is a static variable defined inside a function, which must not conflict with other static variables even if they have the same name.  One easy way to do that is to add a random number to the end of the variable to make it unique.  A similar thing can happen as a result of optimization passes that duplicate functions or merge functions together.

```
rohan:2009$ cat tmp.c
extern int sub2 (int *);
int sub0 (void) { static int i; sub2 (&i); return i; }
int sub1 (void) { static int i; sub2 (&i); return i; }
rohan:2010$ gcc -O2 -c tmp.c
rohan:2011$ nm tmp.o
                 U _GLOBAL_OFFSET_TABLE_
0000000000000004 b i.1797
0000000000000000 b i.1801
0000000000000000 T sub0
0000000000000020 T sub1
                 U sub2
rohan:2012$ 
```

So we now have two i variables, but their names have been modified to make them unique.  The debug info, if you use -g, will map the link time names for the symbols to their original source file names.

bg2d, 2019-09-30:
That makes sense. Thank you!

closed 2019-09-30 by bg2d